zephyrproject-rtos/zephyr #20072: Incompatible pointer types in Nordic Driver nrfx_usbd.h

solsbarry, 2019-10-23:
**Describe the bug**
I added the config option 
`CONFIG_NEWLIB_LIBC=y`
To my prj.conf file, and I get 13 warning from this file:
`zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c`
The warnings are all 
`passing argument 1 of 'atomic_and' from incompatible pointer type [-Wincompatible-pointer-types]`
`expected 'atomic_t *' {aka 'int *'} but argument is of type 'uint32_t *' {aka 'long unsigned int *'}`
Line 926:
`(void)(NRFX_ATOMIC_FETCH_AND(&m_ep_dma_waiting, ~(1U << ep2bit(ep))));`
is one of the offending lines.
The issue is that the config option  (which I need) changed uint32_t from type unsigned int to type long unsigned int. On my platform those are the same but the problem is that the compiler issues a warning when you should have a int and instead give it a long int. 

if you change line 272 from
`static uint32_t m_ep_dma_waiting;`
to 
`static unsigned int m_ep_dma_waiting;`
Then the warnings go away.

or you could use u32_t which is probably better because it comes from zephyr and so this should be type consistent with the rest of zephyr's code. This could be a larger issue where Nordic should be using zephyr types instead of standard library types all over. 

**To Reproduce**
Steps to reproduce the behavior:
prj.conf file with
CONFIG_USB=y
CONFIG_NEWLIB_LIBC=y

1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056
3. make
4. See error

**Expected behavior**
I would expect no compile warning.

**Impact**
There is not actual problem because long int and int are the same, but the 13 warnings are annoying. 

**Environment (please complete the following information):**
 - Linux
 - Toolchain: Arm GNU 
 - V1.14.99 from the the NCS 1.0.0 release

emob-nordic, 2019-10-25:
Hello. The issue You described was know problem in zephyr upstream, but  it's resolved in zephyr 2.0.
I've checked against current master in ncs with zephyr 2.0 and it works fine.
For now I would suggest the hack You did, but in the future the bug would be gone together with ncs relase including zephyr 2.0

aescolar, 2019-11-19:
If this bug is indeed fixed in master, this bug does would not apply anymore. Therefore I close. If the bug is present in master please reopen it.

closed 2019-11-19 by aescolar