zephyrproject-rtos/zephyr #20109: altera_nios2 support decision required

pabigot, 2019-10-24:
Due to lack of support the [gpio driver for altera_nios2](https://github.com/zephyrproject-rtos/zephyr/issues/18530) is expected to be removed in about two weeks when 2.1 completes feature freeze.

It seems like that removing the GPIO driver will means the target platform should also be removed.  Probably sooner rather than later.

What's going to be done about this?

dleach02, 2019-10-24:
@carlescufi , @galak

andrewboie, 2019-10-24:
We had the nios 2 arch port in tree for a considerable time before the GPIO driver appeared. Why not just drop the driver instead of the entire arch port?

pabigot, 2019-10-24:
Most of the boards I've dealt with need GPIO for one thing or another, but if it's unnecessary for this platform that'd be fine.  I just didn't want to get to the point where the driver suddenly disappears and nobody was ready for the fallout.

andrewboie, 2019-10-24:
I think removing the entire arch port will not decrease the fallout of removing the GPIO driver for Nios II.

pabigot, 2019-10-29:
A note for any TSC discussion: the same issue applies to the `cc2650` platform (for which recent interest has been expressed).  Can it stay (without GPIO support), or should it go?

Unlike (apparently) nios2 this is real hardware, and I'd be pretty annoyed if I saw it was listed as a supported platform but then later found out I couldn't use the buttons or LEDs.

(And the `pcal9535a` external GPIO chip, which may be less important, is also slated for removal without deprecation.)

andrewboie, 2019-10-29:
> Unlike (apparently) nios2 this is real hardware,

The Nios II port runs on the Altera Max10 board, which unless I'm hallucinating is real hardware

> I'd be pretty annoyed if I saw it was listed as a supported platform but then later found out I couldn't use the buttons or LEDs.

I really can't understand where you're coming from. Removing the platform completely is somehow better than supporting the platform with the caveat that they might have to write a GPIO driver??

nashif, 2019-10-29:
fixed in #20214

stephanosio, 2019-10-29:
> It seems like that removing the GPIO driver will means the target platform should also be removed.

Please don't remove NIOS II support. Removing a platform support just because GPIO driver is not available makes no sense, especially for NIOS II.

If you are using NIOS II (i.e. a soft-processor for ~Altera~Intel FPGAs), it is very likely that you don't even have a PIO core instantiated to begin with, as that is just extra LEs wasted and you really don't need it most of the times.

pabigot, 2019-10-29:
> I really can't understand where you're coming from. Removing the platform completely is somehow better than supporting the platform with the caveat that they might have to write a GPIO driver??

For the maker community and others who are just interested in using Zephyr without having to take on fixing its functional gaps, I think a case can be made that the answer is yes: it's better to not have it than mislead people about its usability.

Is there no minimal bar for what it means to support a platform?  blinky working would seem to be a reasonable minimal-support feature.

dleach02, 2019-10-29:
I believe the discussion we had re: GPIO was to remove GPIO support from boards that did not have a maintainer that could do the work on the new GPIO API. If someone comes along and needs GPIO support for these platforms then they can do the work to enable the new driver support. 

I get that a platform may be extremely less useful without GPIO and completely removing the platform may be the next step. It all  comes down to having people that are actually supporting the platforms and avoiding getting blocked an larger changes from platforms that don't have support..

nashif, 2019-11-08:
the driver was removed.

closed 2019-11-08 by nashif