zephyrproject-rtos/zephyr #20110: Crash in hci_driver.c when create_connection_cancel is issued after create connection

ahmukunda, 2019-10-24:
PLATFORM: NRF52832, sample project hci_uart
Commit: 566d07e00cce33f70ddc759d383950b9600c217b

HCI_LE_Create_Connection_Cancel command crashes pointing to hci_driver.c if sent after HCI_LE_Create_Connection command. 
If HCI_LE_Create_Connection_Cancel is sent without sending HCI_LE_Create_Connection command, it executes as expected.

Execute the following commands to reproduce the behavior:
1. Send_HCI_Reset
2. Send_HCI_EXT_SetBDADDRCmd
3. Send_HCI_Set_Event_Mask
4. Send_HCI_LE_Set_Event_Mask
5. Send_HCI_LE_Create_Connection
6. Send_HCI_LE_Create_Connection_Cancel

**Expected behavior**
Send_HCI_LE_Create_Connection_Cancel should be executed correctly after executing Send_HCI_LE_Create_Connection.

**Impact**
Showstopper

Log:
[Log.txt](https://github.com/zephyrproject-rtos/zephyr/files/3767561/Log.txt)

aescolar, 2019-10-24:
CC @cvinayak @carlescufi

carlescufi, 2019-10-24:
We are hitting this assert: https://github.com/zephyrproject-rtos/zephyr/blob/6c34983ed22fefcb3dea4c8c7b24ba9e50fc6d68/subsys/bluetooth/controller/hci/hci_driver.c#L209

able-eldhose-oyo, 2019-10-24:
After applying this patch, I could see a LE_Connection_Complete_Event after issuing HCI_LE_Create_Connection_Cancel.
Is this an expected behavior?

```
Packet "HCI_LE_Generic_Report_Event", Opcode 0x003e
Parameters:
Event Opcode : 0x01 (LE Connection Complete Event)
Status : 0x02 (Unknown Connection Identifier)
Connection Id : 0x0000
Role : 0x00 (Link Role Master)
Peer Address Type : 0x00 (Public)
Peer Address : "00:00:00:00:00:00"
Connection Interval : 0x0000
Connection Latency : 0x0000
Supervision Timeout : 0x0000
Clock Accuracy : 0x00
```

joerchan, 2019-10-24:
> After applying this patch, I could see a LE_Connection_Complete_Event after issuing HCI_LE_Create_Connection_Cancel.
> Is this an expected behavior?
>
> ```
> Packet "HCI_LE_Generic_Report_Event", Opcode [...]

Just checked, this seems correct.

From the BT specification:
```
If the cancellation was successful then, after the HCI_Command_Complete
event for the HCI_LE_Create_Connection_Cancel command, either an LE
Connection Complete or an HCI_LE_Enhanced_Connection_Complete event
shall be generated. In either case, the event shall be sent with the error code
Unknown Connection Identifier (0x02)
```

PR #20117 (2019-10-24): Bluetooth: hci: Fix Create Connection Cancel assert
Since the LE (Enhanced) Connection Complete that comes as part of
cancelling a connection does not come through the priority rx thread in
hci_driver.c, the event class was not being calculated and cached.
Hence, calculate and cache the class whenever an event is received as
part of invoking a command.

Fixes #20110.
commit 8c2b849a (2019-10-24): Bluetooth: hci: Fix Create Connection Cancel assert

Since the LE (Enhanced) Connection Complete that comes as part of
cancelling a connection does not come through the priority rx thread in
hci_driver.c, the event class was not being calculated and cached.
Hence, calculate and cache the class whenever an event is received as
part of invoking a command.

Fixes #20110.
closed 2019-10-24 by carlescufi