zephyrproject-rtos/zephyr #20118: Zephyr BLE stack to work on the CC2650 Launchpad

asmakhalil, 2019-10-24:
Hey!

I am trying to change the current zephyr bluetooth implementation of the cc2650 sensor-tag to work for the cc2650 launchpad as they both have the same bluetooth chip (cc2650)

I would like to get some guidance on the changes I have to make in order to get bluetooth to work on my launchpad in order to get the chip on it advertising - I don't care about any other functionality, namely how to alter zephyr sensortag cc2650 code to be platform agnostic in terms of bluetooth features.

I have started by changing the board.h file on the sensor tag to match that of the launchpad and yet I havent been able to get bluetooth working on my chip. Debugging has indicated a problem with bt hci driver not being registered. I checked and it turned out that the kconfig script was not setting a value to BT_UART_ON_DEV_NAME, I fixed that to be "zephyr,bt-uart" and yet it's still not beaconing.

Does anyone have any info on what I could do to get it to work on the cc2650 that's on the launchpad?

Thanks!

Asma

jhedberg, 2019-10-24:
Are you sure Bluetooth is supported for the sensor-tag? I don't see any hints of that in its board definition (besides a reference that that HW supports it, which is not the same as Zephyr support being available). @carlescufi any idea?

asmakhalil, 2019-10-24:
Hey!

Thanks for your answer, I can see it as a supported board here: https://docs.zephyrproject.org/latest/boards/arm/cc2650_sensortag/doc/index.html. Is that not the same as bluetooth being supported on the cc2650 by zephyr? If not, what boards does zephyr support in terms of bluetooth?

Thanks!

Asma

jhedberg, 2019-10-24:
I'm not familiar with this board, but there's a "Supported features" section on that page which doesn't include Bluetooth.

You can get some idea of which boards support Bluetooth by issuing the following git command in the tree: `git grep "default BT" boards/`. Generally, the Nordic Semiconductor-based boards are the safest bet since currently they are the only ones we have "full stack" support for, i.e. both controller and host.

asmakhalil, 2019-10-24:
Oh! i missed that part. Thanks! Well i guess my question is how much work would go into porting bluetooth to work with a board that has a cc26xx ble chip on it?

stephanosio, 2019-10-25:
@asmakhalil CC26XX is a strange beast; it has an RF offloading processor (Cortex-M0) that is separate from the main processor and its firmware source code is not publicly available. Since the RF baseband hardware can only be accessed by this processor, Zephyr will have to appease it somehow.

While the interface for controlling the RF offloading processor (called "Doorbell") is documented in the Technical Reference Manual, the actual BLE driver implementation using it is only available in compiled binary form. Given that the public documentation is by no means complete enough to provide all the information for necessary for implementing the driver, you will have to do some reverse engineering to get the "baseband" driver implemented.

In summary, it will be a lot of work without direct support from TI.

carlescufi, 2019-10-25:
@asmakhalil unless you can provide a binary build of the Controller (Link Layer) that runs on the M0 and then you write a driver for the main application processor similar to what STM has done here: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/bluetooth/hci/ipm_stm32wb.c then I don't see how we can proceed.

bwitherspoon, 2019-10-25:
I did some work with CC2650 sensor tag and the CC13x0 / CC26x0 series. All the drivers for the CC13x2 / CC26x2 including the 802.15.4 driver will work with the CC2650 if you add the CC26x0 SDK and make some minor adjustments. Some of this work can be found in my CC13x0 / CC26x0 branch: https://github.com/bwitherspoon/zephyr/commits/cc13x0_cc26x0. There is no bluetooth support for the CC13x2 / CC26x0 series either though.

I agree with @stephanosio commits. The 802.15.4 driver was written with the using the doorbell commands documented in the TRM. Getting bluetooth working will be a lot more effort and I often found the TRM lacking details.

The main problem you are going to have is the limited amount of memory on the the CC13x0 / CC26x0. I was able to get the 802.15.4 driver, a UART driver, a minimal networking stack and had almost no room for an application. I have doubts that the bluetooth subsystem would fit at all.

cfriedt, 2020-02-11:
I'm working on adding support for the CC13x0 / CC26x0 under PR #21631 with the Split-Link-Layer API Just a few steps away from having a working implementation, although I would say there is definitely room for improvement, outside of some general improvement mentioned in #13519.

closed 2020-10-04 by github-actions[bot]