zephyrproject-rtos/zephyr #20122: Deadlock in ASAN leak detection on exit

vanwinkeljan, 2019-10-24:
**Describe the bug**

Running sanitycheck with address sanitizer (ASAN) enabled could lead to deadlock on native_posix_64 board. The deadlock occurs during program exit and if the leak sanitizer (LSAN) functionality is disabled the deadlock does not occur. 

Note that the problem does not occur on native_posix board as LSAN is not supported on i386.

**To Reproduce**

Based on PR https://github.com/zephyrproject-rtos/zephyr/pull/19093

Steps to reproduce the behavior:
1. sanitycheck -N -p native_posix_64 --enable-asan --enable-lsan
2. wait for deadlock

**Expected behavior**
No deadlock occurs 

**Impact**
LSAN functionality can not be used in CI run

**Environment**
 - OS: Linux
 - Toolchain: Host (gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0)
 - Version: PR https://github.com/zephyrproject-rtos/zephyr/pull/19093 with as base c1ea45280bffc4a43037a56893a6644bc6520e46

Note that deadlock is also seen in CI runs

**Additional context**

_GDB Debug info:_
```
(gdb) i th
  Id   Target Id                                                Frame
* 1    Thread 0x7f32e9966780 (LWP 24999) "zephyr.exe"           0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  2    Thread 0x7f32e42fe700 (LWP 25002) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  3    Thread 0x7f32e3afd700 (LWP 25003) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  4    Thread 0x7f32e32fc700 (LWP 25004) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  5    Thread 0x7f32e4aff700 (LWP 25005) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  6    Thread 0x7f32e2afb700 (LWP 25006) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  7    Thread 0x7f32e22fa700 (LWP 25007) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  8    Thread 0x7f32e1af9700 (LWP 25008) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  9    Thread 0x7f32e12f8700 (LWP 25009) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  10   Thread 0x7f32e0af7700 (LWP 25010) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  11   Thread 0x7f32e02f6700 (LWP 25011) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  12   Thread 0x7f32dfaf5700 (LWP 25012) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  13   Thread 0x7f32df2f4700 (LWP 25013) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  14   Thread 0x7f32deaf3700 (LWP 25014) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  15   Thread 0x7f32de2f2700 (LWP 25015) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
  16   Thread 0x7f32ddaf1700 (LWP 25016) "zephyr.exe" (Exiting) 0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
(gdb) bt
#0  0x00007f32e88b9eee in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
#1  0x00007f32e88c842c in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
#2  0x00007f32e88c8758 in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
#3  0x00007f32e7e51615 in __cxa_finalize (d=0x7f32e8b0f420) at cxa_finalize.c:83
#4  0x00007f32e87dce03 in ?? () from /usr/lib/x86_64-linux-gnu/libasan.so.4
#5  0x00007ffe29e7ceb0 in ?? ()
#6  0x00007f32e9787b73 in _dl_fini () at dl-fini.c:138
Backtrace stopped: frame did not save the PC
(gdb)
```

vanwinkeljan, 2019-10-25:
Deadlock does not occur when using LLVM/CLANG instead of GCC.

Tested with PR #19093 + PR #19589 with as base c1ea452.
Test cmd: `ZEPHYR_TOOLCHAIN_VARIANT=llvm sanitycheck -N -p native_posix_64 --enable-asan --enable-lsan`

aescolar, 2019-10-25:
@vanwinkeljan Which particular sample/test is crashing?

vanwinkeljan, 2019-10-25:
> @vanwinkeljan Which particular sample/test is crashing?

Unfortunately it is a moving target and it is not always the same test that hangs.
In my last runs it was `tests/net/traffic_class/net.traffic_class.rx_5_tx_7`

Before you look further in to this issue I would like to try with static linked lib ASAN, this is the main difference between llvm and gcc (assuming it is not a plain bug in the GCC LSAN implementation).

vanwinkeljan, 2019-10-25:
Running sanitycheck with static linked GCC libasan still result in a deadlock but results in a more useful stack trace

```
(gdb) attach 21872
Attaching to program: sanity-out/native_posix_64/tests/bluetooth/gatt/bluetooth.gatt/zephyr/zephyr.exe, process 21872
[New LWP 21875]
[New LWP 21876]
[New LWP 21877]
[New LWP 21882]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
(gdb) bt
#0  0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
#1  0x0000564af2a5c53c in __lsan::CheckForLeaks() [clone .part.29] ()
#2  0x0000564af2a5c868 in __lsan::DoLeakCheck() ()
#3  0x00007fe4c3a04615 in __cxa_finalize (d=0x564af2ce28a0) at cxa_finalize.c:83
#4  0x0000564af2976673 in __do_global_dtors_aux ()
#5  0x00007ffc54d5caa0 in ?? ()
#6  0x00007fe4c49a3b73 in _dl_fini () at dl-fini.c:138
Backtrace stopped: frame did not save the PC
(gdb) i th
  Id   Target Id                                                Frame
* 1    Thread 0x7fe4c4b82780 (LWP 21872) "zephyr.exe"           0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
  2    Thread 0x7fe4c02fe700 (LWP 21875) "zephyr.exe" (Exiting) 0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
  3    Thread 0x7fe4bfafd700 (LWP 21876) "zephyr.exe" (Exiting) 0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
  4    Thread 0x7fe4c0aff700 (LWP 21877) "zephyr.exe" (Exiting) 0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
  5    Thread 0x7fe4bf2fc700 (LWP 21882) "zephyr.exe" (Exiting) 0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
(gdb) thread 2
[Switching to thread 2 (Thread 0x7fe4c02fe700 (LWP 21875))]
#0  0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
(gdb) bt
#0  0x0000564af2a4264e in __sanitizer::BlockingMutex::Lock() ()
#1  0x0000564af2a57c55 in __sanitizer::ThreadRegistry::FinishThread(unsigned int) ()
#2  0x0000564af2a40d97 in __asan::AsanThread::Destroy() ()
#3  0x00007fe4c3fd0408 in __nptl_deallocate_tsd () at pthread_create.c:300
#4  0x00007fe4c3fd181b in __nptl_deallocate_tsd () at ../sysdeps/nptl/futex-internal.h:200
#5  start_thread (arg=0x7fe4c02fe700) at pthread_create.c:473
#6  0x00007fe4c3ae288f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

vanwinkeljan, 2019-10-25:
Tested with static linking against `gcc-9 (Ubuntu 9.2.1-9ubuntu2~18.04.1) 9.2.1 20191008` and issue still occurs. Looks like a bug in the GCC ASAN implementation, only short term workaround I see is to shutdown threads before calling exit.

vanwinkeljan, 2019-10-25:
Alternative solution to this issue would be to change the compiler to `llvm` if ASAN is enabled.

aescolar, 2019-10-26:
> Alternative solution to this issue would be to change the compiler to llvm if ASAN is enabled.

We cannot add this kind of dependencies. We could add a note in the Kconfig option warning people about deadlocks with gcc and ASAN, but not prevent people from using it or force them to use something else.

vanwinkeljan, 2019-10-27:
@aescolar A workaround for this problem could be to disable _at exit_ leak checking for GCC with the code snippet below, this still leaves the option to the user to enable it by setting it in the ASAN_OPTIONS environment variable `ASAN_OPTIONS="leak_check_at_exit=1"`.

**Code Snippet:**
```
const char *__asan_default_options() {
       return
#if defined(CONFIG_64BIT) && defined(__GNUC__) && !defined(__clang__)
               /* Running leak detection at exit could lead to a deadlock on
                * 64-bit boards if GCC is used.
                * https://github.com/zephyrproject-rtos/zephyr/issues/20122
                * */
               "leak_check_at_exit=0:"
#endif
               "";
}
```

vanwinkeljan, 2019-10-27:
Looks like running ` __lsan_do_leak_check` in the `ON_EXIT` native task handler does not deadlock LSAN (at least locally for me).

vanwinkeljan, 2019-11-05:
Possibly related GCC bug https://gcc.gnu.org/bugzilla/show_bug.cgi?id=88791

PR #20320 (2019-11-05): Disable LSAN "at exit" leak checking when building for 64-bit targets with GCC
Disable "at exit" memory leak check by LSAN if building for a 64-bit target with GCC.
This is need to fix a potential deadlock in GCCs libasan implementation.

Fixes #20122

These commits where separated from PR #19093
commit 4286cb83 (2019-11-05): debug: Disable "at exit" LSAN leak check

Disable "at exit" memory leak check by LSAN if building for a 64-bit
target with GCC. This is need to fix a potential deadlock in GCCs
libasan implementation.

Fixes #20122
closed 2019-11-05 by andrewboie