zephyrproject-rtos/zephyr #20125: Add system call to enter low power mode and reduce latency for deep sleep entry

albertofloyd, 2019-10-24:
This enhancement is related to long latency observed to enter deep sleep.

While default power management policy is flexible enough to allow to optimize power consumption in active mode and deep sleep, the time to reach the latter is longer that some apps may require. 

As an example, the sequence to enter deep sleep start with app either suspending all threads all these waiting for an external event to let the idle thread to run.
Once the idle thread runs, it does call _sys_suspend  with the amount of ticks that system can be idle, this is followed by checks against the supported the sleep states and 

This overall takes 500ms up-to 2s, which is longer than expected when Zephyr-enabled devices are connected to systems that may request an almost immediate suspend ~300ms

A possible solution to reduce latency will be to expose a system call that app can use to force a determined power state, or similar function for apps to trigger a request for low power entry.
The solution should not be mutually exclusive with existing active power policy

**Describe alternatives you've considered**
Calling _sys_suspend within the main app thread reduces the latency a bit, but there are cases where the request competes with the one from idle thread.

wentongwu, 2020-01-08:
the basic idea to fix this high latency is to start a dedicated and very high priority work_q which is used to handle the force suspend request, and it will never compete with the suspend handle in idle thread. But that requires the device drivers implement DEVICE_PM_FORCE_SUSPEND_STATE state. @pabigot @vanti

albertofloyd, 2020-01-09:
@wentongwu would it be possible to have this "bypass/force" request to be handled by SoC 
for boards/soc that do not support DEVICE_POWER_MANAGEMENT ? 
If DEVICE_POWER_MANAGEMENT is support perform the suspension as you indicated and if not to rely on soc sys_set_power_state() is updating updating HW blocks in low power mode?

wentongwu, 2020-01-10:
@albertofloyd Implementing DEVICE_PM_FORCE_SUSPEND_STATE state is to make sure the possible device ongoing transfers stop and store the states. It depends on SOC not OS where we put that functional code, just to make sure there is no HW issue. 
By the way do you have any requirements for the deep sleep request interface between Host and Zephyr, like when received request, Zephyr have to give response for it, etc.

albertofloyd, 2020-01-16:
Understood, I meant to ask if force suspend request Soc requirements can be specified like this.
1) Requires the SoC device drivers to implement DEVICE_PM_FORCE_SUSPEND_STATE state
or
2) Requires SoC power management module to ensure HW transitions
Original comment makes me wonder if 1) is mandatory or 2) is good enough before 1) is in place

wentongwu, 2020-01-16:
> Understood, I meant to ask if force suspend request Soc requirements can be specified like this.
> 
> 1. Requires the SoC device drivers to implement DEVICE_PM_FORCE_SUSPEND_STATE state
>    or
> 2. Requires SoC power management module to ensure HW handles the force suspend if drivers do not support suspend state.
>    Your initial comment sounds like 1) is mandatory, Is that a hard requirement or if 2) is implemented should be good enough

@albertofloyd I think ensure HW handles the force suspend is good enough.

PR #21881 (2020-01-25): power: add system power management direct trigger mode
Add system power management direct trigger mode. In this mode
application thread can directly put system in sleep or deep
sleep mode instead of waiting for idle thread to do it, so that
it can reduce latency to enter low power mode.

Fixes: #20125.
closed 2020-01-25 by nashif