zephyrproject-rtos/zephyr #20135: LL.TS LL/CON/SLA/BV-11-C fails with MD fix

mtpr-ot, 2019-10-25:
The LL.TS.5.1.1 test case LL/CON/SLA/BV-11-C [Slave Sending Termination] fails in BabbleSim  after the commit fcd9612016e170abc4abb5387a2b3057ea840859, which correctly prevents MD bit being set indiscriminately.
This is caused by the master not acknowleding the TERMINATION_IND, now  the slave is no longer sending it with MD=1.

**To Reproduce**
Steps to reproduce the behavior:
1. Run test case LL/CON/SLA/BV-11-C, with nRF52 board as Upper- and Lower Tester
2. Test fails with disconnect cause **0x22** (BT_HCI_ERR_LL_RESP_TIMEOUT) instead of **0x16** (BT_HCI_ERR_LOCALHOST_TERM_CONN)
3. Inspecting the air trace shows that master did not acknowledge the TERMINATION_IND

**Expected behavior**
Slave (IUT) should disconnect with **0x16** (BT_HCI_ERR_LOCALHOST_TERM_CONN)

**Impact**
This issue, besides failing LL verification test, slows down BLE disconnection

cvinayak, 2019-10-29:
@mtpr-ot This test is passing on the EBQ Qualifier, sent logs by email.

mtpr-ot, 2019-10-29:
> @mtpr-ot This test is passing on the EBQ Qualifier, sent logs by email.

Logs recevied and look fine, thanks. However, the problem is that if you run it twice without reset, it fails at my desk.
Looking int the implementation in ull_conn.c, it looks like "lll.master.terminate_ack" is never set to zero, so after first terminate ack, it cleans up before sending the ack.
@cvinayak, can you confirm?

cvinayak, 2019-10-29:
@mtpr-ot confirmed. I will try sending a PR tomorrow.

PR #20233 (2019-11-04): Bluetooth: controller: split: Fix missing terminate_ack initialization
Add the missing initialization of master role terminate_ack
flag. This caused slave initiated remote termination as
procedure timeout.

Fixes #20135.
commit eed51cc0 (2019-11-04): Bluetooth: controller: split: Fix missing terminate_ack initialization

Add the missing initialization of master role terminate_ack
flag. This caused slave initiated remote termination as
procedure timeout.

Fixes #20135.
closed 2019-11-04 by carlescufi