zephyrproject-rtos/zephyr #20136: kernel: undefined reference with --no-gc-sections

masz-nordic, 2019-10-25:
**Describe the bug**
When building with `--no-gc-sections` option to ld, reference to `sys_rand32_get` is undefined.
This is because its implementation is only included when one of:
- CONFIG_TIMER_RANDOM_GENERATOR
- CONFIG_X86_TSC_RANDOM_GENERATOR
- CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR
- CONFIG_XOROSHIRO_RANDOM_GENERATOR

is enabled, even though the function is called disregarding any of those configs.

**To Reproduce**
1. Add:
```diff
--- a/samples/basic/blinky/CMakeLists.txt
+++ b/samples/basic/blinky/CMakeLists.txt
@@ -3,5 +3,6 @@
 cmake_minimum_required(VERSION 3.13.1)
 include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
 project(blinky)
-
+zephyr_ld_options(-Wl,--no-gc-sections)
+zephyr_ld_options(-Wl,--orphan-handling=place)
 target_sources(app PRIVATE src/main.c)
```
2. Build blinky:

```
west build -b nrf52840_pca10056
```

**Expected behavior**
Sample builds.

**Impact**
Annoyance.

**Screenshots or console output**
```
zephyr/kernel/libkernel.a(init.c.obj): in function `z_early_boot_rand32_get':
/home/marcin/zephyr/zephyr/kernel/init.c:466: undefined reference to `sys_rand32_get'
```

**Environment:**
 - OS: Linux 5.2.21-1-MANJARO
 - Toolchain: Zephyr SDK 0.10.3
 - 55471982f68fed13671353585609507d2096bade with modification above

aescolar, 2019-10-25:
@masz-nordic before filling more bug reports like this ones, maybe you would like to fill a question if these are actually accepted as bugs?
Zephyr has been compiled garbage collecting in the linker for very long time. So you are doing something which is not really expected, and find something which is not exactly a bug, as that code which references those functions is not actually used.

masz-nordic, 2019-10-25:
@aescolar 
No need to add more bug reports - there are only 2 places in Zephyr that use an undefined reference (in my builds).
No garbage collecting just shows the bugs that can be reproduced in specific circumstances. For example in this case by enabling `CONFIG_STACK_CANARIES`. It then fails in the same way. So yes, I do think this is a bug.

SebastianBoe, 2019-10-25:
Note that using --no-gc-sections is a valid use-case.

Sometimes a function is in use without the linker being able to detect the usage.

andrewboie, 2020-02-15:
> Note that using --no-gc-sections is a valid use-case.

Is it?

To me this is an unsupported configuration.

ceolin, 2020-02-15:
it is not being breaking the build because this code is only called if STACK_CANARIES or STACK_POINTER_RANDOM is enabled, and these options implicitly enable a random source.
If any other code unconditionally call it we will have a problem. I'm just ifdefing  this function to be called if some of that options is enabled.

SebastianBoe, 2020-02-17:
> Is it?
> 
> To me this is an unsupported configuration.

Yes. Because sometimes a function is in use without the linker realizing it, in these cases we should not garbage collect unused sections. An example of when this is happens is when an application supports RPC. Then procedures might be called only externally without any internal references to the procedures.

ceolin, 2020-03-16:
Both cases were fixed. Just closing this issue.

closed 2020-03-16 by ceolin