zephyrproject-rtos/zephyr #20140: CMake: syscall macro's are not generated for out of tree DTS_ROOT

Laczen, 2019-10-25:
**Describe the bug**
When using a custom device tree with:
```
west build -b <board name> -- -DDTS_ROOT=<path to dts root>
```
include files that are under the `DTS_ROOT/include` are not added to the syscall_macros list.

**To Reproduce**
1. Create a out of tree devicetree with a header that uses __syscall declarations.
2. Use above command to build
3. The build command shows the error:`#include <syscalls/your_header.h> no such file or directory`

**Expected behavior**
It should be possible to add header that uses __syscall declarations in a out of tree devicetree.

**Impact**
The behavior makes it impossible to use out of tree dts.

SebastianBoe, 2019-10-25:
Hi, added support for this in https://github.com/zephyrproject-rtos/zephyr/pull/7328 , but at that time it was believed that we should not support out-of-tree declarations of syscalls.

Perhaps @andrewboie or @nashif could comment on if things have changed and we now want to support this or not.

> Driver system calls are declared at the subsystem level and not the actual drivers themselves.
> I do not see a use-case for having out-of-tree driver subsystems or out of tree system calls. You should still be able to create out-of-tree drivers without making new syscalls.

-- @andrewboie

Laczen, 2019-10-25:
@nashif, @andrewboie, supporting out-of-tree declarations of syscalls allows development of subsystems to take place before it is integrated into zephyr. This can avoid disturbances inside the tree, only when the development is mature enough it could be added to the tree. I think there are other use cases as well, e.g. development of subsystems that will never go into the tree because of licensing, ownership, ...

andrewboie, 2019-10-25:
I won't stand in the way if you choose to resurrect that patch.

SebastianBoe, 2019-10-29:
Will get this supported then.

closed 2020-01-22 by nashif