zephyrproject-rtos/zephyr #20161: Facing issue to setup zephyr on ubuntu

jagannath-sahoo, 2019-10-25:
I'm getting this error

CONTRIBUTING.rst    kernel          scripts     zephyr-env.sh
jagannath@jagannath-Lenovo-G500:~/zephyrproject/zephyr$ west build -b qemu_x86 samples/hello_world
-- west build: build configuration:
       source directory: /home/jagannath/zephyrproject/zephyr/samples/hello_world
       build directory: /home/jagannath/zephyrproject/zephyr/build
       BOARD: qemu_x86 (origin: command line)
-- west build: generating a build system
Zephyr version: 2.0.99
CMake Warning at /home/jagannath/zephyrproject/zephyr/cmake/app/boilerplate.cmake:160 (message):
  The build directory must be cleaned pristinely when changing boards
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Selected BOARD reel_board
CMake Error at /home/jagannath/zephyrproject/zephyr/cmake/toolchain/zephyr/host-tools.cmake:26 (message):
  The file
  '/home/jagannath/Documents/RTOS/toolchain/xtensa-lx106-elf/bin:/home/jagannath/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/jagannath/jdk1.8.0/bin/:/home/jagannath/node-v10.15.3/bin/:/home/jagannath/arduino/arduino-1.8.9:/home/jagannath/flutter/flutter/bin:/home/jagannath/qemu/gcc-arm/bin:/home/jagannath/zephyr-sdk-0.10.3/sdk_version'
  was not found.  Is
  ZEPHYR_SDK_INSTALL_DIR=/home/jagannath/Documents/RTOS/toolchain/xtensa-lx106-elf/bin:/home/jagannath/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/jagannath/jdk1.8.0/bin/:/home/jagannath/node-v10.15.3/bin/:/home/jagannath/arduino/arduino-1.8.9:/home/jagannath/flutter/flutter/bin:/home/jagannath/qemu/gcc-arm/bin:/home/jagannath/zephyr-sdk-0.10.3
  misconfigured?
Call Stack (most recent call first):
  /home/jagannath/zephyrproject/zephyr/cmake/host-tools.cmake:3 (include)
  /home/jagannath/zephyrproject/zephyr/cmake/app/boilerplate.cmake:443 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
ERROR: command exited with status 1: /home/jagannath/.local/bin/cmake -B/home/jagannath/zephyrproject/zephyr/build -S/home/jagannath/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=qemu_x86
Any suggestion what to do?

dbkinder, 2019-10-25:
The hint here is likely:

    The build directory must be cleaned pristinely when changing boards

It looks like you tried building for a different board and then switched to using qemu.  Cmake caches board and sample app information so if you change either of those in a subsequent build, you need to make that cache go away.
Try  changing the west command to add the “-p” option for cleaning out the build folder (or just delete the build/ folder):

     west build --p b qemu_x86 samples/hello_world

-- david

nashif, 2020-01-20:
question answered.

closed 2020-01-20 by nashif