zephyrproject-rtos/zephyr #20163: doc: storage settings not clear

sdalu, 2019-10-25:
It is very hard to understand(didn't succeed) current documentation about storage settings, some terms are not explained and documentation is not in sync with code

doc: `settings_src_register(&settings_custom_itf);`
code: `settings_src_register(&zephyr_settings_fw_store);` (peripheral_dis)

what is the meaning of prefix/suffix:
* csi
* itf

what is the relation between store and itf?

more examples in sample directory would be welcome

nvlsianpu, 2019-10-29:
@fnde-ot As you implemented custom back-ends support, can you put some light on this questions?

nvlsianpu, 2019-10-29:
As I can tell `settings_src_register()` role is to register handler required to serve the back-end. As multiple backnd might be used as setting origin therefore structure with handlers is provided in container for source list node.
itf - probably stand for `interface`
csi - probably stand for `custom interface`
Anyway. I wouldn't take care about what exactly above acronyms stand for.

fnde-ot, 2019-10-30:
The documentation is actually wrong, as we need to pass a **settings_store** node to settings_src_register and settings_dst_register APIs, and not a **settings_store_itf**. Good catch!

**settings_store** is a linked list node containing a **settings_store_itf** element (see include/settings/settings.h).

I will update the doc. Meanwhile you can checkout examples of registering a backend in settings_fcb.c, settings_file.c, and settings_nvs.c (custom backends are no different).

PR #20231 (2019-10-31): settings: fix custom backend example
Fix example of custom backend implementation

Fixes: #20163.
commit 468a834a (2019-10-31): settings: fix custom backend example

Fix example of custom backend implementation

Fixes: #20163.
closed 2019-10-31 by ioannisg