zephyrproject-rtos/zephyr #20172: devicetree support for compound elements

pabigot, 2019-10-27:
I need a devicetree property as described below (documentation detail removed):

```
  dpd-wakeup-sequence:
    type: compound
    required: false
    description: Specifies wakeup durations for devices without RDPD
```

with an example use:

    dpd-wakeup-sequence = <30000 20 45000>;

The value of this is an array of three integers.  `binding-template.yaml` specifies that `compound` remains the catch-all for more complex types, but when that's used all the generated macros include deprecation markers.

The documentation separately mentions `type: array` (which is not a listed available type) which works.

What is the appropriate type to use for this?  What would be appropriate if the element types were heterogeneous (what `compound` used to mean)?

In either case the generation needs to produce the property value as an initializer sequence, rather than individual elements, so some tooling enhancement will be required.

ulfalizer, 2019-10-27:
`type: array` not being documented together with the other types is a (bad) oversight. Will fix tomorrow.

ulfalizer, 2019-10-28:
Maybe a `length:` field could be added to bindings for checking the length too, if needed.

Something like

```yaml
properties:
    dpd-wakeup-sequence:
        type: array
        length: 3
        required: false
```

pabigot, 2019-10-28:
Yeah, that could be useful, as long as its absence means any number of entries is allowed.

To be clear: `array` is intended to replace all uses of `compound`, including heterogenous structures?  Or does `compound` still get used for that.  You can answer by making it clear in the PR that improves the documentation.

pabigot, 2019-12-10:
Closed by #20173.  (The answer to the last question is that `array` means a list of integer values only.  `compound` appears to be used for "not an array or a phandle-array".)

closed 2019-12-10 by pabigot