zephyrproject-rtos/zephyr #20176: tests/drivers/pwm/pwm_api failed on reel_board.

chen-png, 2019-10-28:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=reel_board tests/drivers/pwm/pwm_api
2. west flash
3. See error

**Screenshots or console output**
****** Booting Zephyr OS build zephyr-v2.0.0-1547-gda5a595f248f *****
Running test suite pwm_basic_test

starting test - test_pwm_usec
[PWM]: 0, [period]: 2000, [pulse]: 1000
E: PWM pin 0 not enabled through DTS configuration.
Fail to set the period and pulse width

    Assertion failed at ZEPHYR_BASE/tests/drivers/pwm/pwm_api/src/test_pwm.c:112: test_pwm_usec: test_task(DEFAULT_PWM_PORT, DEFAULT_PERIOD_USEC, DEFAULT_PULSE_USEC, UNIT_USECS) == TC_PASS is false

FAIL - test_pwm_usec

starting test - test_pwm_nsec
[PWM]: 0, [period]: 2000000, [pulse]: 1000000
E: PWM pin 0 not enabled through DTS configuration.
Fail to set the period and pulse width

    Assertion failed at ZEPHYR_BASE/tests/drivers/pwm/pwm_api/src/test_pwm.c:130: test_pwm_nsec: test_task(DEFAULT_PWM_PORT, DEFAULT_PERIOD_NSEC, DEFAULT_PULSE_NSEC, UNIT_NSECS) == TC_PASS is false

FAIL - test_pwm_nsec

starting test - test_pwm_cycle
[PWM]: 0, [period]: 64000, [pulse]: 32000
E: PWM pin 0 not enabled through DTS configuration.
Fail to set the period and pulse width

    Assertion failed at ZEPHYR_BASE/tests/drivers/pwm/pwm_api/src/test_pwm.c:148: test_pwm_cycle: test_task(DEFAULT_PWM_PORT, DEFAULT_PERIOD_CYCLE, DEFAULT_PULSE_CYCLE, UNIT_CYCLES) == TC_PASS is false

FAIL - test_pwm_cycle

Test suite pwm_basic_test failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: da5a595f

ioannisg, 2019-11-27:
Hi @jfischer-phytec-iot any updates on this?

nashif, 2019-11-28:
This test was not customized for reelboard, it does need board specific settings AFAICT. I doubt it worked before...

jfischer-phytec-iot, 2019-11-28:
I do not think it worked for any nRF based board.

```
boards/arm/nrf52_pca10040/nrf52_pca10040.dts:156:       ch0-pin = <17>;
boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev.dts:117:   ch0-pin = <22>;
boards/arm/nrf52833_pca10100/nrf52833_pca10100.dts:166: ch0-pin = <13>;
boards/arm/nrf5340_dk_nrf5340/nrf5340_dk_nrf5340_cpuapp_common.dts:105: ch0-pin = <28>;
boards/arm/nrf52832_mdk/nrf52832_mdk.dts:98:    ch0-pin = <22>;
boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts:166: ch0-pin = <13>;
boards/arm/holyiot_yj16019/holyiot_yj16019.dts:54:      ch0-pin = <29>;
boards/arm/nrf52840_mdk/nrf52840_mdk.dts:100:   ch0-pin = <22>;
boards/arm/nrf52811_pca10056/nrf52811_pca10056.dts:119: ch0-pin = <13>;
boards/arm/bl654_dvk/bl654_dvk.dts:114: ch0-pin = <13>;
boards/arm/actinius_icarus/actinius_icarus_common.dts:119:      ch0-pin = <10>;
boards/arm/bl652_dvk/bl652_dvk.dts:90:  ch0-pin = <17>;
boards/arm/nrf9160_pca10090/nrf9160_pca10090_common.dts:123:    ch0-pin = <2>;
boards/arm/nrf52840_papyr/nrf52840_papyr.dts:99:        ch0-pin = <13>;
boards/arm/reel_board/dts/reel_board.dtsi:137:  ch0-pin = <13>;
```

PR #21067 (2019-11-28): test: pwm_api: fix test for nRF based boards
Fix pwm_api test for nRF based boards.

Fixes: #20176
commit 04d962b0 (2019-11-28): test: pwm_api: fix test for nRF based boards

Fix pwm_api test for nRF based boards.

Fixes: #20176
closed 2019-11-28 by nashif