zephyrproject-rtos/zephyr #20205: ztest testing.ztest does not have a prj.conf with CONFIG_ZTEST=y

hakehuang, 2019-10-29:
**Describe the bug**
if we directy build the tests/ztest/base it will report
zephyr/tests/ztest/base/src/main.c:7:10: fatal error: ztest.h: No such file or directory

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
build clean

**Impact**
unknown

**Screenshots or console output**

```
zephyr/tests/ztest/base/src/main.c:7:10: fatal error: ztest.h: No such file or directory
 #include <ztest.h>
          ^~~~~~~~~
compilation terminated.
CMakeFiles/app.dir/build.make:62: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (Zephyr SDK)
 - v1.14.1

hakehuang, 2019-10-29:
@MaureenHelm can I assign it to me and fix it?

aescolar, 2019-11-04:
@hakehuang (Risking stepping in somebody's toes) Everybody is welcome to submit a bugfix PR

hakehuang, 2019-11-11:
Thanks @maureen, pull request create #20568

stephanosio, 2019-11-11:
@hakehuang This is not a bug. It is quite hard to follow without a sufficient amount of background knowledge on this, but see below:

https://github.com/zephyrproject-rtos/zephyr/blob/d1d603afc3558f31a4728d70179d1cb5d74890c5/tests/ztest/base/testcase.yaml#L2-L4

https://github.com/zephyrproject-rtos/zephyr/blob/0297756342fb60e6a64813aac046f4d241cbd369/scripts/sanity_chk/boards/unit/unit_testing/unit_testing.yaml#L1-L6

https://github.com/zephyrproject-rtos/zephyr/blob/0297756342fb60e6a64813aac046f4d241cbd369/tests/ztest/base/CMakeLists.txt#L4-L7

https://github.com/zephyrproject-rtos/zephyr/blob/0297756342fb60e6a64813aac046f4d241cbd369/subsys/testsuite/unittest.cmake#L30-L31

hakehuang, 2019-11-12:
@stephanosio , in this setting users can only build with sanitycheck, and can not work with directly steps like:
cmake ..
make
or 
west build

stephanosio, 2019-11-12:
@hakehuang This is not meant to be built manually. If you want to build it manually without sanitycheck, you must do `cmake -DBOARD=unit_testing ..` as sanitycheck automatically does.

stephanosio, 2019-11-12:
```
stephanos@DOMINUS:~/Dev/zephyr-public/zephyr/tests/ztest/base/build$ cmake -DBOARD=unit_testing ..
CMake Deprecation Warning at /home/stephanos/Dev/zephyr-public/zephyr/subsys/testsuite/unittest.cmake:4 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:7 (include)

-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/stephanos/Dev/zephyr-public/zephyr/tests/ztest/base/build
stephanos@DOMINUS:~/Dev/zephyr-public/zephyr/tests/ztest/base/build$ make -j32
Scanning dependencies of target kobj_types_h_target
[ 20%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[ 20%] Built target kobj_types_h_target
Scanning dependencies of target testbinary
[ 40%] Building C object CMakeFiles/testbinary.dir/src/main.c.o
[ 60%] Building C object CMakeFiles/testbinary.dir/home/stephanos/Dev/zephyr-public/zephyr/subsys/testsuite/ztest/src/ztest.c.o
[ 80%] Building C object CMakeFiles/testbinary.dir/home/stephanos/Dev/zephyr-public/zephyr/subsys/testsuite/ztest/src/ztest_mock.c.o
[100%] Linking C executable testbinary
[100%] Built target testbinary
stephanos@DOMINUS:~/Dev/zephyr-public/zephyr/tests/ztest/base/build$ ls -l testbinary
-rwxrwxr-x 1 stephanos stephanos 13232 Nov 12 11:36 testbinary
stephanos@DOMINUS:~/Dev/zephyr-public/zephyr/tests/ztest/base/build$ ./testbinary
Running test suite framework_tests
===================================================================
starting test - test_empty_test
PASS - test_empty_test
===================================================================
starting test - test_assert_tests
PASS - test_assert_tests
===================================================================
starting test - test_assert_mem_equal
PASS - test_assert_mem_equal
===================================================================
Test suite framework_tests succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
stephanos@DOMINUS:~/Dev/zephyr-public/zephyr/tests/ztest/base/build$
```

hakehuang, 2019-11-12:
in this case, how can we port to other platform, without add a prj.conf? @stephanosio

stephanosio, 2019-11-12:
> in this case, how can we port to other platform, without add a prj.conf? @stephanosio

@hakehuang This is a special test that tests the test framework itself on the host machine. It is not meant to run on any other platform.

hakehuang, 2019-11-12:
@stephanosio thanks for the detailed explaination, can we add platform_whitelsit instead of tag: unit, so that it is more generanal unstanding why?

stephanosio, 2019-11-12:
> can we add platform_whitelsit instead of tag: unit

If you are referring to `type: unit`, no. There are many special treatments for this, so unless you are planning to do an overhaul (in which case, you should talk to @nashif, by the way), it would be better left as is.

For instance:
https://github.com/zephyrproject-rtos/zephyr/blob/0297756342fb60e6a64813aac046f4d241cbd369/scripts/sanitycheck#L1941-L1943

hakehuang, 2019-11-12:
ok, thanks

hakehuang, 2019-11-12:
not an issue

closed 2019-11-12 by hakehuang