zephyrproject-rtos/zephyr #20217: Extend qemu_cortex_r5 test coverage

stephanosio, 2019-10-29:
At the time of writing, the `qemu_cortex_r5` board based on `xilinx_zynqmp` SoC is only able to run very basic non-multitasking samples and tests (e.g. `hello_world` sample).

In order to improve test coverage on the `qemu_cortex_r5` platform and facilitate CI testing of the ARM Cortex-R port, QEMU needs to be patched to provide a better emulation of the RPU of the Xilinx ZynqMP SoC; at the same time, incorrect `xilinx_zynqmp` SoC implementation on Zephyr needs to be fixed.

## QEMU Tasks
- [ ] PATCH: arm/xlnx-zynqmp: put APUs and RPUs in separate CPU clusters
(https://patchwork.kernel.org/patch/10717697)
- [ ] PATCH: xlnx-zynqmp: add the timers
(https://patchwork.kernel.org/patch/10048347)
- [ ] PATCH: xlnx-zynqmp: add PL390 GIC for RPU
(WIP)

A pull request incorporating the above patches is to be made on zephyr_rtos/qemu. As for the new patch(es), a pull request will be made on Xilinx/qemu and, hopefully, Xilinx will submit an upstream patch.

## Zephyr
- [ ] Implement PL390 GICv1 support (try to refactor the existing GIC-400 GICv2 code)
- [ ] Fix incorrect xilinx_zynqmp dts (the current dts used by RPU is really that of APU, especially the GIC-400 mapping that only applies to APU)

## Note
- Since the current Zephyr architecture cannot support AMP of Zynq APUs and RPUs in one project, we should consider the APU and RPU to be different hardware platforms (i.e. consider renaming `xilinx_zynqmp` to `xilinx_zynqmp_rpu`; when Cortex-A support is added in the future, add `xilinx_zynqmp_apu`; `qemu_cortex_a53` should then utilise `xilinx_zynqmp_apu`).

stephanosio, 2019-10-29:
cc @bbolen @wjliang @carlocaione @ioannisg

stephanosio, 2019-10-29:
Sample qemu RPU boot command line using the latest dts:
`qemu-system-aarch64 -machine arm-generic-fdt -nographic -dtb ~/Dev/xilinx-qemu-devicetrees/build/zcu102-arm.dtb -serial mon:stdio -device loader,file=/home/stephanos/Dev/zephyr-public/zephyr/tests/kernel/timer/timer_monotonic/build_cr5/zephyr/zephyr.elf,cpu-num=4 -device loader,addr=0xff5e023c,data=0x80008fde,data-len=4 -device loader,addr=0xff9a0000,data=0x80000218,data-len=4`

bwasim, 2019-12-15:
I'm trying out Zephyr and an emulator is my obvious first choice.. I've been able to build and execute Zephyr on QEMU-R5 but I am unable to debug the app / kernel (while starting Qemu machine, I tell the GDB to listen to a port for TCP connections and then connect it via an IDE).. I think the problem is that the R5 image is ARM while the Qemu model is Aarch64, and so this incompatibility generates problems (See https://forums.xilinx.com/t5/Embedded-Linux/Running-application-on-R5-of-zcu102-xilinx-QEMU-model/td-p/905515).. Have any of you guys been able to debug this ? If so, can you kindly list down the steps.. 

--Bilal

stephanosio, 2019-12-15:
@bwasim I have never tried debugging Zynq RPU under QEMU, but the Xilinx docs mention that a special gdb build is required for this (i.e. multi-arch build configured with `--enable-targets=all`), which is not included in the Zephyr SDK at the moment.

https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841606/QEMU+-+Zynq+UltraScalePlus#QEMU-ZynqUltraScalePlus-DebuggingtheA53sandtheR5sfromGDB

I will look into this tomorrow and see if I can get this special gdb build included in the Zephyr SDK.

bwasim, 2019-12-15:
Thanks for the update.. 

On a side note, I'm currently trying to get a functional board support for ZCU102-R5 (physical board) with support for Ethernet enabled (I don't see the GEM IP ported to Zephyr yet).. Will open a PR for it soon (Thanks to your previous work, I already have the UART / Kernel up and running).. 

--Bilal

stephanosio, 2019-12-16:
@bwasim Just tested Xilinx QEMU RPU debugging, and it works with a gdb that is configured with `--target=aarch64-zephyr-elf --enable-targets=arm-zephyr-eabi` (you can just use `gdb-multiarch` that is commonly available too).

I will try to get this included for the Zephyr SDK 0.11.0 release.

bwasim, 2019-12-16:
Very cool.. Thanks..

andrewboie, 2019-12-19:
What debugger is that?

stephanosio, 2019-12-19:
> What debugger is that?

@andrewboie It is gdb with the gdb-dashboard.

https://github.com/cyrus-and/gdb-dashboard

stephanosio, 2020-01-07:
All PRs merged.

closed 2020-01-07 by stephanosio

wentongwu, 2020-02-25:
@stephanosio not familiar with this platform, but what's the CPU clock frequency for this QEMU, thanks

stephanosio, 2020-02-25:
> @stephanosio not familiar with this platform, but what's the CPU clock frequency for this QEMU, thanks

@wentongwu In terms of QEMU, it is meaningless to talk about the CPU clock frequency for obvious reasons.

When running on real hardware (ZCU102), the clock frequency would depend on various configurations (e.g. on-board oscillator frequency, PLL configuration, ... handled by the PMU).

wentongwu, 2020-02-25:
yes, for normal mode we don't need that, but I'm trying to enable QEMU icount mode for qemu_cortex_r5, and it need CPU clock frequency to some math, if any interest, please take a look https://github.com/zephyrproject-rtos/zephyr/pull/22904 and https://github.com/zephyrproject-rtos/zephyr/issues/14173 . Thanks