zephyrproject-rtos/zephyr #20232: Bluetooth: Kernel panic on gatt discover in shell app

cvinayak, 2019-10-30:
**Describe the bug**
Performing the cmd ```gatt discover``` in the tests/bluetooth/shell application before discover completes causes kernel panic.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 ../tests/bluetooth/shell/
3. ninja
4. ninja flash

Flash onto two DKs. And open two serial port terminals (ex: minicom -D /dev/ttyACMx)

Commands on DK1:
```
uart:~$ bt init
uart:~$ bt oob
```
pick the bdaddr to be used from DK2
```
uart:~$ bt advertise on
```

Commands on DK2:
```
uart:~$ bt init
uart:~$ bt connect  4f:ec:da:2a:2c:d9 (random)
```
Wait for auto connection update to be complete

```
bt conn-update 320 320 9 800
```

Commands on DK1:
```
uart:~$ gatt discover
uart:~$ gatt discover
```

The following assertion is seen:
```
uart:~$ Discover complete
uartA:~$ SERTION FAIL [req->func] @ ZEPHYR_BASE/subsys/bluetooth/host/att.c:301
[00:01:14.702,026] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000012d  r2/a3:  0x00000001
[00:01:14.702,026] <err> os: r3/a4:  0x0003086c r12/ip:  0x00000070 r14/lr:  0x0000fc4d
[00:01:14.702,056] <err> os:  xpsr:  0x61000000
[00:01:14.702,056] <err> os: Faulting instruction address (r15/pc): 0x000277d2
[00:01:14.702,056] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic
[00:01:14.702,087] <err> os: Current thread: 0x20002284 (sysworkq)
[00:01:14.751,342] <err> os: Halting system
```

**Expected behavior**
Either the second call to gatt discover failed or was queued for future execution.

**Impact**
showstopper

**Screenshots or console output**
[minicom_slave.cap.txt](https://github.com/zephyrproject-rtos/zephyr/files/3788093/minicom_slave.cap.txt)
[minicom_master.cap.txt](https://github.com/zephyrproject-rtos/zephyr/files/3788095/minicom_master.cap.txt)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb
 - Commit SHA or Version used: 5a601721e6

**Additional context**
Notice the use of large connection interval and slave latency.

joerchan, 2019-10-30:
This is documented for `bt_gatt_discover`.
```
 *  Note: This procedure is asynchronous therefore the parameters need to
 *  remains valid while it is active.
```

So this is only a problem that the shell allows the second call to discover.

Vudentz, 2019-10-30:
Yeah, we might be able to prevent that to happen by checking if the parameters are already in use.

joerchan, 2019-10-30:
@Vudentz I'll just submit a PR with some mutexes.

PR #20239 (2019-11-04): Bluetooth: Shell: Disallow GATT commands when params are in use
GATT parameters need to remain valid while the procedure is active.
Disallow GATT commands while the procedure is active to avoid asserting.

Fixes: #20232
commit 23e8e689 (2019-11-04): Bluetooth: Shell: Disallow GATT commands when params are in use

GATT parameters need to remain valid while the procedure is active.
Disallow GATT commands while the procedure is active to avoid asserting.

Fixes: #20232
closed 2019-11-04 by jhedberg