zephyrproject-rtos/zephyr #20236: usb: api: Cleanup of current inclusion path for USB

emob-nordic, 2019-10-30:
## Introduction
USB public API should be reworked so user can see only what is necessary.
### Problem description
User app shouldn't be aware of what is happening in the stack and in device controller.
Currently in the public API [usb_device.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h) some additional structures, enums and callbacks are defined which should not bother user. For example:
- [endpoint callback](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L102)
- [usb setup packet structure](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L85)
- [usb config data](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L172).
- many others

Additionally usb_device.h [includes](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L39) usb_dc.h file, and by that users can see unnecessary declarations which should not be visible by them.

If we want users to use the classes and USB stack in proper way, We should avoid exposing unnecessary functionalities to them.

If anyone wants to use the USB devices directly and declare its own class then shall achieve that by including [usb_dc.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/usb/usb_dc.h)

What is more. Current implementation creates unnecessary chaos. For example, have look at endpoint callback.
- one declared here [usb_device.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L102)
- one declared here [usb_dc.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/usb/usb_dc.h#L119)

Both are exaclty the same, but someone once added this because he could not see it directly.

### Proposed change

- I suggest all mentioned structures, enum, callbacks, etc should be placed in [usb_dc.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/drivers/usb/usb_dc.h) as this API actually interacts with devices directly.

- Clean up API inclusion paths as now usb_device.h includes usb_dc.h. This should be done exactly the other way.

## Detailed RFC

### Proposed change (Detailed)

I came across this problem when facing point 1 from #13403 and initially my idea was to propose solution together with this PR #20044, however I've already started this work and it seems a big api change so I wont put this in one PR.

### Dependencies

I don't see any but I might be missing something. This is strict USB topic IMO.

### Concerns and Unresolved Questions

I count on You as I don't see any right now. Please feel free to comment.

## Alternatives

??

carlescufi, 2019-11-07:
@emob-nordic this seems like a change worth pursuing. Are you going to send a PR that addresses this particular issue? I think we should indeed do that.

emob-nordic, 2019-11-07:
@carlescufi Yes, I am. I already have something prepared but its not clean enough to be candidate for PR.

carlescufi, 2019-11-07:
> @carlescufi Yes, I am. I already have something prepared but its not clean enough to be candidate for PR. I will continue for that for sure.

Great, thanks Emil. I really think this is worthwhile pursuing.