zephyrproject-rtos/zephyr #20246: Module Request: hal_unisoc

Mani-Sadhasivam, 2019-10-30:
This RFC issue is targeted towards adding Unisoc HAL libraries as a module to zephyr. Currently, there is only UWP5 HAL is available here: https://github.com/unisoc/hal_uwp5.git

This HAL library will be used by SoCs such as UWP5661 SoC in 96Boards IVY5661 board. The rest of the HALs for different family SoCs from Unisoc can also be hosted in this module in future!

Preferred repository/module name: hal_unisoc

Once the module is added, I'll open a PR for adding the UWP5661 SoC and IVY5661 board support.

Mani-Sadhasivam, 2019-10-31:
tagging @galak @MaureenHelm @carlescufi

carlescufi, 2019-11-07:
@Mani-Sadhasivam are you still waiting for this module? I can create the repo

Mani-Sadhasivam, 2019-11-07:
@carlescufi Yes, please. I'm blocked.

carlescufi, 2019-11-07:
https://github.com/zephyrproject-rtos/hal_unisoc

closed 2019-11-07 by carlescufi