zephyrproject-rtos/zephyr #20269: Add support for opamps in MCUs

NidhCypress, 2019-10-31:
Many MCUs have integrated opamps. Please add peripheral API support for opamps.

henrikbrixandersen, 2021-04-12:
Do you have an example of a SoC with built-in opamp?

towen, 2021-08-07:
+1 for this. 
For example STM32G4xx as seen on the STM32 B-G431-esc1 motor control application demo.
https://www.st.com/en/evaluation-tools/b-g431b-esc1.html

closed 2022-01-12 by github-actions[bot]

KozhinovAlexander, 2025-01-06:
> Many MCUs have integrated opamps. Please add peripheral API support for opamps. 
> 

You mentioned "many" MCUs. So far I know stm32g4 has OpAmps. Could you please mention other manufacturer MCUs with OpAmps?
It is important to know, since at least basic device tree pattern shall cover as much manufacturers as possible.

warasilapm, 2025-01-09:
A great number of the STM32 lineup now has integrated opamps, including F3, G4, L4, U5, and even U0. A broader range also have comparators. Definitely a complex feature set to generalize, though.

nono313, 2025-03-10:
Some EFM32 chips seem to have opamp peripherals as well ([https://www.silabs.com/documents/public/application-notes/an0038.1-efm32-series-1-operational-amplifiers.pdf](https://www.silabs.com/documents/public/application-notes/an0038.1-efm32-series-1-operational-amplifiers.pdf)).

Is there a minimum number of SoC required to provide a generic driver API ?

KozhinovAlexander, 2025-03-10:
> Some EFM32 chips seem to have opamp peripherals as well [...]

I think, there is no min number os socs

Spencer-Sawyer, 2025-06-09:
Fwiw I came to find this because the MCXA156 has an opamp and the MCXN9 has 3.

ZhaoxiangJin, 2025-08-04:
I have introduced OPAMP device driver API https://github.com/zephyrproject-rtos/zephyr/pull/94040 in Zephyr, if you are still interested, please comment and provide your feedback. Thanks.