zephyrproject-rtos/zephyr #20274: Kconfig new libc changes cause echo server cmake error

rmstoi, 2019-11-01:
After libc config changes

9555f82d28
0e59f6846f
c1f865b412

echo server cmake fails:

```
cd samples/net/sockets/echo_server/build
cmake  -DBOARD=native_posix -DCONF_FILE=prj.conf  ..
```

Results in cmake error:
```
warning: the choice symbol NEWLIB_LIBC (defined at lib/libc/Kconfig:29) is selected by the following symbols, but select/imply has no effect on choice symbols
 - MIPI_SYST_LIB (defined at modules/Kconfig.syst:4)

Error: Aborting due to non-whitelisted Kconfig warning 'warning: the choice symbol NEWLIB_LIBC
(defined at lib/libc/Kconfig:29) is selected by the following symbols, but select/imply has no
effect on choice symbols  - MIPI_SYST_LIB (defined at modules/Kconfig.syst:4)'. Note: If this
warning doesn't point to an actual problem, you can add it to the whitelist at the top of
zephyrproject/zephyr/scripts/kconfig/kconfig.py.

CMake Error at zephyrproject/zephyr/cmake/kconfig.cmake:210 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  zephyrproject/zephyr/cmake/app/boilerplate.cmake:461 (include)
  CMakeLists.txt:5 (include)

```

SebastianBoe, 2019-11-01:
I believe master is broken because dd9480913e9 is incompatible with the libc changes, but these two commits were not integration tested in CI.

rmstoi, 2019-11-01:
Reverting:

9555f82
0e59f68
c1f865b

Fixes cmake issues in my environment

PR #20277 (2019-11-01): libc: Use select REQUIRES_FULL_LIBC instead of select NEWLIB_LIBC
Changed select NEWLIB_LIBC to select REQUIRES_FULL_LIBC.

This fixes #20274
closed 2019-11-01 by galak