zephyrproject-rtos/zephyr #20285: ST lis2dh sample with motion callback

olsky, 2019-11-01:
Trying to set TH and DUR for the TRIGGER mode, it seems any value passed is triggering motion callback. 

The functions are there but no sample -- could this be covered?

avisconti, 2019-11-04:
@olsky 
What do you mean any value?
Can you be more specific with some example values?

Concerning the sample we can add it.

olsky, 2019-11-04:
Thanks, sample would be great!
I’m trying to set wake/sleep functionality, apparently choosing wrong values.

I should post some logging — let me collect and share...

Btw: what is the right way to set a trigger say when having 4g / 8g acceleration?

olsky, 2019-11-04:
here is a log from my test app, I also uploaded a tiny sample app.

---

SPM: NS image at 0xc000
SPM: NS MSP at 0x20020a90
SPM: NS reset vector at 0xdc8d
SPM: prepare to jump to Non-Secure image.

[00:00:00.016,571] [1B][0m<inf> lis2dh: int1 on pin=0 cfg=0x26[1B][0m
[00:00:00.024,139] [1B][0m<inf> lis2dh: int2 on pin=1 cfg=0x26[1B][0m
[00:00:00.032,470] [1B][0m<inf> lis2dh: bus=I2C_2 fs=2, odr=0x1 lp_en=0x0 scale=597[1B][0m

***** Booting Zephyr OS build v1.14.99-ncs3-snapshot2-2647-gd6e67554cfeb *****

lis2dh app test starting...

[00:00:00.056,182] [1B][0m<dbg> lis2dh.lis2dh_start_trigger_int1: ctrl1=0x17 @tick=1841[1B][0m

lis2dh_acc_slope_config SENSOR_ATTR_SLOPE_TH
lis2dh_acc_slope_config SENSOR_ATTR_SLOPE_TH > LIS2DH_REG_CTRL4 = 0
lis2dh_acc_slope_config SENSOR_ATTR_SLOPE_TH > range_g = 2
lis2dh_acc_slope_config SENSOR_ATTR_SLOPE_TH > slope_th_ums2 = 64

[00:00:00.095,153] [1B][0m<inf> lis2dh: int2_ths=0x0 range_g=2 ums2=63[1B][0m

lis2dh_acc_slope_config SENSOR_ATTR_SLOPE_DUR

[00:00:00.109,222] [1B][0m<inf> lis2dh: int2_dur=0x4[1B][0m
[00:00:00.116,180] [1B][0m<inf> lis2dh: status=0xf x=1408 y=0 z=16384[1B][0m

( x y z ) = ( 0.840576  0.000000  9.781248 )

[00:00:01.130,798] [1B][0m<inf> lis2dh: status=0xf x=1472 y=-64 z=16384[1B][0m

( x y z ) = ( 0.878784  -0.038208  9.781248 )

[00:00:02.142,333] [1B][0m<inf> lis2dh: status=0xf x=1344 y=0 z=16512[1B][0m

( x y z ) = ( 0.802368  0.000000  9.857664 )

[00:00:03.153,625] [1B][0m<inf> lis2dh: status=0xf x=1536 y=0 z=16384[1B][0m

( x y z ) = ( 0.916992  0.000000  9.781248 )

[00:00:04.164,886] [1B][0m<inf> lis2dh: status=0xf x=1280 y=0 z=16320[1B][0m

( x y z ) = ( 0.764160  0.000000  9.743040 )

[00:00:05.176,147] [1B][0m<inf> lis2dh: status=0xf x=1600 y=128 z=15872[1B][0m

( x y z ) = ( 0.955200  0.076416  9.475584 )

lis2dh_handler_motion ~~> ##################### MOTION
[00:00:06.147,491] [1B][0m<inf> lis2dh: status=0xf x=1344 y=0 z=16256[1B][0m

( x y z ) = ( 0.802368  0.000000  9.704832 )

[20191104-lis2dh12.zip](https://github.com/zephyrproject-rtos/zephyr/files/3805512/20191104-lis2dh12.zip)

avisconti, 2019-11-07:
> Btw: what is the right way to set a trigger say when having 4g / 8g acceleration?

So, you are interested into setting the lis2dh12 (driver is lis2dh anyway) in order to trigger the interrupt only above a certain threshold, correct?

In your code I see:

		val_threshold.val1 = 0;
		val_threshold.val2 = 64;
		err = sensor_attr_set(dev_lis2dh, SENSOR_CHAN_ACCEL_XYZ, SENSOR_ATTR_SLOPE_TH, &val_threshold);

I think this is wrong. I'm not sure this part of the driver has been really tested, but what you have to do is to transform the 4g acceleration threshold (for example) in ms2. 
So 4g == 39.2266 ms2

Then you have to put the integer part in val1 (val1 = 39) and the fractional part multiplied by 10^6 in val2 (val2 = 0.2266*10^6 = 226600). 

See documentation here:
https://docs.zephyrproject.org/latest/reference/peripherals/sensor.html

Then, according to if you want a given axis to stay higher or lower this threshold, you should set ZH/ZL or YH/YL or XH/XL. But it seems that it is not possible to set it. Instead INT2 is configured as a ANY MOTION triggering with XH/YH/ZH to '1'. So, if yo set the 4g as above you should trigger INT2n interrupt with any motion above 4g.

Is it clear?

olsky, 2019-11-07:
Thank you for the update! Let me try for the 4G interrupt...
The sensor is lis2dh12 -- yes, and I would need to update the driver... 

Question on wake/sleep: as the code is there, what are the values to try it?
Could you please share some values?

avisconti, 2019-11-07:
> The sensor is lis2dh12 -- yes, and I would need to update the driver...

You can test any motion on INT2 first

avisconti, 2019-11-07:
> Question on wake/sleep: as the code is there, what are the values to try it?
> Could you please share some values?

I'm not a big expert.
You can take a look to wakeup feature in the application note. There are already register settings as an example.
https://www.st.com/resource/en/application_note/cd00290365.pdf

olsky, 2019-11-07:
> > The sensor is lis2dh12 -- yes, and I would need to update the driver...
> 
> You can test any motion on INT2 first

yes! let me test and I will post results...

olsky, 2019-11-07:
> I think this is wrong. I'm not sure this part of the driver has been really tested, but what you have to do is to transform the 4g acceleration threshold (for example) in ms2.
> So 4g == 39.2266 ms2
> 
> Then you have to put the integer part in val1 (val1 = 39) and the fractional part multiplied by 10^6 in val2 (val2 = 0.2266*10^6 = 226600).
> 
> See documentation here:
> https://docs.zephyrproject.org/latest/reference/peripherals/sensor.html
> 
> 
> Is it clear?

It is clear! Thank you for the hint: it works! :)
I set 2G and 4G and it works... I have to understand more the relation between SENSOR_ATTR_SLOPE_DUR and ODR... but will read the application note.

So first: acceleration as you suggested is ok.

Now, I suppose the code for this driver is very basic and not all the features are implemented. like: wake/sleep --> right?

avisconti, 2019-11-11:
> Now, I suppose the code for this driver is very basic and not all the features are implemented. like: wake/sleep --> right?

Correct.
It seems this driver is very simple. It was written by someone else. Concerning the triggers seems that they have decided to use INT2 for implementing ANY MOTION (in x, y, z). This I think primarily because there is no way to pass the  XH/XL, YH/YH, ZH/ZL information.

avisconti, 2019-11-11:
May I close this bug?

olsky, 2019-11-12:
> Correct.
> It seems this driver is very simple. It was written by someone else. Concerning the triggers seems that they have decided to use INT2 for implementing ANY MOTION (in x, y, z). This I think [...]

Thank you, all clear.

One more question: Could you suggest the right way to implement these features in zephyr? The "classic way" of setting i2c register is feasible... but the interface of zephyr is totally new and need some guidance to make it....

olsky, 2019-11-12:
> May I close this bug?

yes! thank you for your support! 

Btw: any chance ST are making some contribution to zephyr sensors ? :)

avisconti, 2019-11-12:
> > May I close this bug?
> 
> yes! thank you for your support!
> 

You are welcome!

> Btw: any chance ST are making some contribution to zephyr sensors ? :)

Well, At the momenti I'm the only one that is supporting it.
Other people, like @erwango, are supporting the STM32 mcu side.

avisconti, 2019-11-12:
> One more question: Could you suggest the right way to implement these features in zephyr? The "classic way" of setting i2c register is feasible... but the interface of zephyr is totally new and need some guidance to make it....

Mmmh. Don't know.
Maybe you can introduce a specific sensor attribute (SENSOR_ATTR_...) to pass the XH/YH/ZH stuff.

closed 2019-11-12 by avisconti