zephyrproject-rtos/zephyr #20286: West build fails for ESP32 while cmake build works !!

satyanraj, 2019-11-02:
West build fails for ESP32 while cmake build works fine !!
Please find the log below:

Any clues? 
Thanks in advance!

satyan@Satyan-HP-ProBook-440-G2:~/zephyrproject/zephyr$ west build -b esp32 samples/hello_world
-- west build: build configuration:
       source directory: /home/satyan/zephyrproject/zephyr/samples/hello_world
       build directory: /home/satyan/zephyrproject/zephyr/build
       BOARD: esp32 (origin: CMakeCache.txt)
-- west build: building application
[1/97] Preparing syscall dependency handling

[16/97] Building ASM object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj 
ccache /home/satyan/zephyr-sdk-0.10.3/xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.0.0-1617-g0c8e7eaeef0a -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/xtensa/esp32 -I../soc/xtensa/esp32/include -I/home/satyan/zephyrproject/modules/hal/esp-idf/zephyr/../components/esp32/include -I/home/satyan/zephyrproject/modules/hal/esp-idf/zephyr/../components/soc/esp32/include -I/home/satyan/zephyr-sdk-0.10.3/xtensa-zephyr-elf/include/arch/include -I/home/satyan/zephyr-sdk-0.10.3/xtensa-zephyr-elf/include -isystem ../lib/libc/minimal/include -isystem /home/satyan/zephyr-sdk-0.10.3/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/8.3.0/include -isystem /home/satyan/zephyr-sdk-0.10.3/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/8.3.0/include-fixed -Os -imacros/home/satyan/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -xassembler-with-cpp -imacros/home/satyan/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/satyan/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/satyan/zephyrproject/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mlongcalls -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj -MF zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj.d -o zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj -c /home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S: Assembler messages:
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:161: Error: unknown opcode or format name 'wsr.lbeg'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:163: Error: unknown opcode or format name 'wsr.lend'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:165: Error: unknown opcode or format name 'wsr.lcount'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lbeg'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lend'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lcount'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lbeg'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lend'
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lcount'
[21/97] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2.c.obj
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /home/satyan/.local/bin/cmake --build /home/satyan/zephyrproject/zephyr/build

satyan@Satyan-HP-ProBook-440-G2:~/zephyrproject/zephyr$ cmake -B build -GNinja -DBOARD=esp32 samples/hello_world
Zephyr version: 2.0.99
-- Selected BOARD esp32
-- Found west: /home/satyan/.local/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/satyan/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts as base
Devicetree configuration written to /home/satyan/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/satyan/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/satyan/zephyrproject/zephyr/build/zephyr/.config'
No change to '/home/satyan/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: /home/satyan/.cache/zephyr
CMake Warning at /home/satyan/zephyrproject/zephyr/boards/common/esp32.board.cmake:7 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/satyan/zephyrproject/zephyr/boards/xtensa/esp32/board.cmake:3 (include)
  /home/satyan/zephyrproject/zephyr/cmake/app/boilerplate.cmake:503 (include)
  CMakeLists.txt:5 (include)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/satyan/zephyrproject/zephyr/build

carlescufi, 2019-11-04:
@satyanraj with CMake you are only generating the build system but not actually building from your log. You are mising:

`ninja -C build`

in your cmake log

mbolivar, 2019-11-04:
```
/home/satyan/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lcount'
```

My guess is this error will appear in the `ninja` output as well. It looks like you've got the wrong toolchain configured for your board.

ExtremeGTX, 2019-11-06:
I think for esp32 you must use espressif toolchain `ZEPHYR_TOOLCHAIN_VARIANT=espressif`
I got similar errors using zephyr toolchain

```
FAILED: zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj 
ccache /src/esp32/zephyr-sdk/xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.0.0-1415-g11faaacd1e22 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/xtensa/esp32 -I../soc/xtensa/esp32/include -I/src/esp32/zephyr-extremegtx/modules/hal/esp-idf/zephyr/../components/esp32/include -I/src/esp32/zephyr-extremegtx/modules/hal/esp-idf/zephyr/../components/soc/esp32/include -I/src/esp32/zephyr-sdk/xtensa-zephyr-elf/include/arch/include -I/src/esp32/zephyr-sdk/xtensa-zephyr-elf/include -isystem ../lib/libc/minimal/include -isystem /src/esp32/zephyr-sdk/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/8.3.0/include -isystem /src/esp32/zephyr-sdk/xtensa-zephyr-elf/bin/../lib/gcc/xtensa-zephyr-elf/8.3.0/include-fixed -Os -imacros/src/esp32/zephyr-extremegtx/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -xassembler-with-cpp -imacros/src/esp32/zephyr-extremegtx/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/src/esp32/zephyr-extremegtx/zephyr/samples/drivers/i2c_scanner=CMAKE_SOURCE_DIR -fmacro-prefix-map=/src/esp32/zephyr-extremegtx/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mlongcalls -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj -MF zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj.d -o zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2-util.S.obj -c /src/esp32/zephyr-extremegtx/zephyr/arch/xtensa/core/xtensa-asm2-util.S
/zephyr/arch/xtensa/core/xtensa-asm2-util.S: Assembler messages:
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:161: Error: unknown opcode or format name 'wsr.lbeg'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:163: Error: unknown opcode or format name 'wsr.lend'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:165: Error: unknown opcode or format name 'wsr.lcount'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lbeg'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lend'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:198: Error: unknown opcode or format name 'rsr.lcount'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lbeg'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lend'
/zephyr/arch/xtensa/core/xtensa-asm2-util.S:237: Error: unknown opcode or format name 'rsr.lcount'
[25/97] Building C object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/xtensa-asm2.c.obj
ninja: build stopped: subcommand failed.

```

mbolivar, 2019-11-06:
Reassigning to @andrewboie since this is a build system issue for esp32, not a west issue

satyanraj, 2019-11-06:
> I think for esp32 you must use espressif toolchain `ZEPHYR_TOOLCHAIN_VARIANT=espressif`
> I got similar errors using zephyr toolchain
>
> ```
> FAILED: [...]

Have exported the ESP TOOLCHAIN and IDF path correctly!!

mbolivar, 2019-11-06:
> @satyanraj with CMake you are only generating the build system but not actually building from your log. You are mising:
> 
> `ninja -C build`
> 
> in your cmake log

You still didn't paste this

andrewboie, 2019-11-06:
> 
> 
> Reassigning to @andrewboie since this is a build system issue for esp32, not a west issue

I'm not the appropriate assignee for this

mbolivar, 2019-11-06:
> I'm not the appropriate assignee for this

CODEOWNERS says you maintain esp32.

mbolivar, 2019-11-06:
Changed the title of this issue to reflect that it's not a west issue

andrewboie, 2019-11-06:
> CODEOWNERS says you maintain esp32.

So it does! I've never seen one of these boards :P We'll get an assignee sorted.

ExtremeGTX, 2019-11-07:
@satyanraj i use espressif toolchain for esp32 and zephyr toolchain for qemu_xtensa

satyanraj, 2019-11-11:
> @satyanraj i use espressif toolchain for esp32 and zephyr toolchain for qemu_xtensa

Hi @ExtremeGTX, I am trying to use ESP IDF and toolchain provided by Espressif.

I have also download the Zephyr SDK (latest) where I can find toolchain for ESP32.

I was looking at the bold log and I just noticed that build is using toolchain provided with Zephyr SDK. Is that something causing the build error?

satyanraj, 2019-11-12:
I am able to build the example and run the application. The issue was because of the installed Zeyphr SDK. 
I don't understand why Zeyphr SDK was used for west build even though I have unset ZEPHYR_SDK_INSTALL_DIR ?

ExtremeGTX, 2019-11-12:
Which toolchain will be used is based on the value of env var `ZEPHYR_TOOLCHAIN_VARIANT`if it is `espressif` or `zephyr`

satyanraj, 2019-11-13:
> Which toolchain will be used is based on the value of env var `ZEPHYR_TOOLCHAIN_VARIANT`if it is `espressif` or `zephyr`

@ExtremeGTX : I am not sure about it!!  As far as I understand, for ESP32 the toolchain remains env. var. should always be ZEPHYR_TOOLCHAIN_VARIANT="espressif"
 but based on the env. var. ESPRESSIF_TOOLCHAIN_PATH="/path/to/xtensa-esp32-elf/" and 
ZEPHYR_SDK_INSTALL_DIR the toolchain gets selected.   
https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html#set-up-build-environment 

Please correct me if I am wrong.

antfobe, 2019-12-05:
> @satyanraj i use espressif toolchain for esp32 and zephyr toolchain for qemu_xtensa

Maybe a bit off topic, but have you ever managed to get any of the samples/net working in qemu_xtensa?

mbolivar, 2020-01-08:
> I am able to build the example and run the application

@andrewboie looks like you might be able to close this one

closed 2020-01-09 by andrewboie