zephyrproject-rtos/zephyr #20297: Bluetooth: can't close bt_driver log output

codingspirit, 2019-11-04:
**Describe the bug**
When set **BT_DEBUG_NONE** and **BT_LOG_LEVEL_OFF**, Log module **bt_driver** still print out "bt_driver: TX interrupt but no pending buffer!".
PS: **LOG_OVERRIDE_LEVEL** was set as 0.

**To Reproduce**
Steps to reproduce the behavior:
1. set **BT_DEBUG_NONE** and **BT_LOG_LEVEL_OFF** in menuconfig
2. build and flash
3. See **bt_driver** output error logs

**Expected behavior**
No bt related log should be printed out no matter it's error, warning or info.

**Impact**
A lot of "bt_driver: TX interrupt but no pending buffer!" messed up terminal.

**Screenshots or console output**
[00:09:31.039,000] <err> bt_driver: TX interrupt but no pending buffer!
[00:09:31.039,000] <err> bt_driver: TX interrupt but no pending buffer!
[00:09:31.274,000] <err> bt_driver: TX interrupt but no pending buffer!
[00:09:31.274,000] <err> bt_driver: TX interrupt but no pending buffer!
[00:09:31.274,000] <err> bt_driver: TX interrupt but no pending buffer!
[00:09:31.275,000] <err> bt_driver: TX interrupt but no pending buffer!

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr version 2.0.99

**Additional context**
If enabled shell, bt log can be turned off by **log disable bt_driver**

codingspirit, 2019-11-04:
I'd like to add additional question here that why bt still functional although those "TX interrupt but no pending buffer" keeping jumping out, it will be great if someone can answer me, thanks

jhedberg, 2019-11-04:
I'd be more concerned about this error than the logging behaviour. It looks like the UART driver on your platform (which is it btw?) is misbehaving, i.e. it's generating interrupts even though `uart_irq_tx_disable()` was called. Btw, you can turn off all logging with `CONFIG_BT_DEBUG_LOG=n` however that will turn it off for all Bluetooth modules, i.e. not just the driver.

codingspirit, 2019-11-05:
@jhedberg Thanks for your quick answer, but I already set `CONFIG_BT_DEBUG_LOG` to `n`, this message still pump out.
```
CONFIG_BT=y
CONFIG_BT_HCI=y
CONFIG_BT_CONN=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_LOG_LEVEL_OFF=y
CONFIG_BT_DEBUG_LOG=n
```

If you are interesting in the UART misbehaving, here are some details:
Issue was happened on STM32L451, it's quite confusing me since BT HCI is still functional. Blow parts is our related dts:

```
&usart3 {
	current-speed = <1000000>;
	label = "UART_BLE";
	status = "okay";
	hw-flow-control;
};
```

As I said it confused me a lot since BT functions works normal, I can successfully `bt_enable` and `bt_le_scan_start`.

closed 2019-11-26 by jhedberg

AndreyDodonov-EH, 2020-06-17:
Hi,
I understand that this issue is about incorrect logging behavior, 
but it is also almost the only result about "TX interrupt but no pending buffer".

@codingspirit  How were you getting this issue? Could you fix it?

We are implement streaming app (audio input to BLE), first several minutes it works fine, but then only this error comes (and not more messages are sent).

codingspirit, 2020-06-25:
> Hi,
> I understand that this issue is about incorrect logging behavior,
> but it is also almost the only result about "TX interrupt but no pending buffer".
>
> @codingspirit How were you getting this [...]

@AndreyDodonov-EH 
Hi, we met this issue when we trying to use zephyr based MCU(STM32) as HCI host. We knew that Log level wasn't root cause, but #20961 did help us to get rid of endless "TX interrupt but no pending buffer!".

AndreyDodonov-EH, 2020-06-25:
@codingspirit 
Thanks for the info! You are right,with this change log becomes configurable.
Were you able to solve your original issue?

P.S. We are using though Nordic Thingy 91 (https://www.nordicsemi.com/Software-and-tools/Prototyping-platforms/Nordic-Thingy-91, both host and controller use zephyr)
This fix partially improved the situation https://github.com/zephyrproject-rtos/zephyr/issues/25904
But not completely, these look also relevant, may be they will also be of interest to you
https://github.com/zephyrproject-rtos/zephyr/pull/25954
https://github.com/zephyrproject-rtos/zephyr/issues/23364

albdom, 2020-09-02:
@codingspirit @AndreyDodonov-EH
I have the same issue using Nordic PCA10090, the BLE stopped with this loop error. Were you able to solve the issue?

PR #20961 (2019-11-26): Bluetooth: Fix BT log level config won't take effect
BT log module should be registered as user configured level instead of always using default log level.

Fixes #20297