zephyrproject-rtos/zephyr #20301: tests/drivers/watchdog/wdt_basic_api failed on sam_e70_xplained board.

chen-png, 2019-11-04:
**Describe the bug**
Running tests/drivers/watchdog/wdt_basic_api test case on sam_e70_xplained board, it will loop to report the message below.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=sam_e70_xplained  tests/drivers/watchdog/wdt_basic_api 
2. west flash
3. See error

**Screenshots or console output**
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: ca5c829a84

aurel32, 2019-11-04:
The problem is likely that the RAM is reset when the watchdog is triggered on the SAM70 MCU. IIRC there is/was a bug opened about that, but I can't find it anymore.

chen-png, 2019-11-05:
> The problem is likely that the RAM is reset when the watchdog is triggered on the SAM70 MCU. IIRC there is/was a bug opened about that, but I can't find it anymore.

do you mean the #13468 and #14968, it seems the same issue.

ioannisg, 2019-11-05:
Looks like duplicate; might want to close it. It's anyways, a LOW priority bug.

ashaannajoy, 2020-02-06:
This issue is observed on microchip mec15xxevb_assy6853  also

jocelyn-li, 2020-02-27:
@albertofloyd is this issue caused by limited RAM resource on microchip mec15xxevb_assy6853?

albertofloyd, 2020-02-27:
@jocelyn-li  there is no limited RAM resource. 
In other reset conditions the bootloader fetches the code again from SPI flash and loads into RAM.
Need confirmation from @scottwcpg to see RAM is preserved during WDT_EVENT since that's what the test code is expecting. If RAM is not preserved in this case the test have to skipped.

chen-png, 2020-02-28:
this test has been skipped on sam_e70_xplained, so closed this.
opened another new issue #23147 to track this issue for mec15xxevb_assy6853 board.

scottwcpg, 2020-02-28:
Hi,
MEC150x Boot-ROM clears all SRAM per security requirements of its secure boot loader feature.
You could write a test  flag into VBAT memory which retains state if the VTR or VBAT power rails stay up.

PR #23534 (2020-03-17): tests: wdt_basic_api: make mec15xxevb_assy6853 build only
The bootloader on mec15xxevb_assy6853 clears the memory on reboot,
which would fail the wdt_basic_api test. Instead of simply exclude
the platform, add a build only test for this board so the watchdog
is being built to avoid build breakage.

Fixes #20301
commit 4b192733 (2020-03-17): tests: wdt_basic_api: make mec15xxevb_assy6853 build only

The bootloader on mec15xxevb_assy6853 clears the memory on reboot,
which would fail the wdt_basic_api test. Instead of simply exclude
the platform, add a build only test for this board so the watchdog
is being built to avoid build breakage.

Fixes #20301
closed 2020-03-17 by nashif