zephyrproject-rtos/zephyr #20310: SDHC : Could not enable SPI clock on nucleo_f091rc

Almog2610, 2019-11-04:
I'm trying to write logs to an SD card that's on my nucleo_f091rc board using the SDHC driver, however when I run my program it crashes.
I switched to **subsys/fs/fat_fs sample** to see if I can run it with my current setup, when I try to run it I get the following error : 
```
[0.000000 0.000000] ***** Booting Zephyr OS build zephyr-v2.0.0-1-gd1bfa605c14e *****
[0.006859 0.006859] [00:00:00.000,000] <err> spi_ll_stm32: Could not enable SPI clock
[0.012961 0.006102] [00:00:00.005,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
[5.025380 5.012419] FATAL: ***** HARD FAULT *****
```

My nucleo_f091rc.overlay file : 
```
&spi2 {
           status = "okay";
           cs-gpios = <&gpiob 12 0>;
        sdhc0: sdhc@0 {
                compatible = "zephyr,mmc-spi-slot";
                reg = <0>;
                status = "okay";
                label = "SDHC0";
                spi-max-frequency = <24000000>;
        };
};
```

My prj.conf file :
```
CONFIG_SPI=y
CONFIG_SPI_2=y
CONFIG_DISK_ACCESS=y
CONFIG_DISK_ACCESS_SDHC=y
CONFIG_DISK_ACCESS_SPI_SDHC=y
CONFIG_DISK_SDHC_VOLUME_NAME="SD"
CONFIG_LOG=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_PRINTK=y

# Serial configuration
CONFIG_SERIAL=y
CONFIG_UART_CONSOLE=n

# RTT configuration
CONFIG_HAS_SEGGER_RTT=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. Load the fat_fs sample
2. Enable SPI2

**Expected behavior**
List all of the directories that on the SD card filesystem

**Impact**
Blocker

**Environment :**
 - OS: Linux, Kubuntu 18.04.3 LTS
 - Toolchain: Zephyr SDK
 - Zephyr version used: 2.0.0 relase

**Things that I have tried:**
- Changing SPI ports
- Changing SPI pins
- Changing the wires on my board
- Configure the GPIO pin to **active low** before running the program

erwango, 2019-11-05:
@Almog2610 the SPI clock init issue is really strange.
Do you have a way to test more basic SPI use case?

erwango, 2019-11-05:
@Almog2610, thanks for reporting the issue.

Good news is that https://github.com/zephyrproject-rtos/zephyr/pull/20354 should fix the reported "Could not enable SPI clock" error.
Bad news is that clock init issue was happening on SPI_1 and this is likely not linked with your hard fault.
EDIT: If HARD FAULT is still present, please raise a dedicated issue so we don't mix up.

PR #20354 (2019-11-05): dt: stm32f0: Fix clock bus for SPI1 and few timers
There is no APB2 bus on stm32f0 series.
What could be found as APB2 in CMSIS files is actually
second group of APB (A.K.A APB1_2).
Fix nodes that are using this wrong reference accorss the series.

Fixes #20310
commit a0da140a (2019-11-05): dt: stm32f0: Fix clock bus for SPI1 and few timers

There is no APB2 bus on stm32f0 series.
What could be found as APB2 in CMSIS files is actually
second group of APB (A.K.A APB1_2).
Fix nodes that are using this wrong reference accorss the series.

Fixes #20310
commit 798e12c2 (2019-11-05): dt: stm32f0: Fix clock bus for SPI1 and few timers

There is no APB2 bus on stm32f0 series.
What could be found as APB2 in CMSIS files is actually
second group of APB (A.K.A APB1_2).
Fix nodes that are using this wrong reference accorss the series.

Fixes #20310
commit 5baed7d4 (2019-11-05): dt: stm32f0: Fix clock bus for SPI1 and few timers

There is no APB2 bus on stm32f0 series.
What could be found as APB2 in CMSIS files is actually
second group of APB (A.K.A APB1_2).
Fix nodes that are using this wrong reference accorss the series.

Fixes #20310
closed 2019-11-05 by MaureenHelm