zephyrproject-rtos/zephyr #20325: samples/drivers/i2c_scanner does not work on STM32 NUCLEO and DISCOVERY boards

frantony, 2019-11-04:
**Describe the bug**

The commit 442df97366416a2779882d185c94e36619a1223a (scripts/dts: Remove alias defines for labels) makes the commit 173d07c6e5e7d98318cdbfc87ab1009a0d389a33 (samples: driver: i2c_scanner: use arduino-defined device if available) unusable.

As a result ``samples/drivers/i2c_scanner`` does not work on STM32 NUCLEO and DISCOVERY boards.

**To Reproduce**
Steps to reproduce the behavior:
1. build ``samples/drivers/i2c_scanner`` zephyr image for any supported STM32 NUCLEO or DISCOVERY board;
2. upload zephyr image into the board;
3. run zephyr;
4. see the error message
```
Starting i2c scanner...
I2C: Device driver not found.
```

frantony, 2019-11-04:
@galak @pabigot 

Please help!

pabigot, 2019-11-04:
There are two things that need to be done.

First, unless there's another solution pending (which @galak might be able to say) all Arduino bus values must switch from using a label to using an alias, as with below.

Second, once that's done the test doesn't work because the ST I2C driver doesn't handle the SMBUS command that's used to detect a NAK on the address for an absent subordinate device.  It reports all addresses as being associated with devices.  That's something that @ydamigos might be able to assist with.

```
diff --git a/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi b/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
index 1f47915822..c6a37a6ffd 100644
--- a/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
+++ b/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
@@ -5,6 +5,9 @@
  */
 
 / {
+       aliases {
+               arduino-i2c = &i2c1;
+       };
        arduino_header: connector {
                compatible = "arduino-header-r3";
                #gpio-cells = <2>;
@@ -34,5 +37,3 @@
                           <21 0 &gpiob 8 0>;   /* D15 */
        };
 };
-
-arduino_i2c: &i2c1 {};
diff --git a/samples/drivers/i2c_scanner/src/main.c b/samples/drivers/i2c_scanner/src/main.c
index 44c1724377..ac71a0d85f 100644
--- a/samples/drivers/i2c_scanner/src/main.c
+++ b/samples/drivers/i2c_scanner/src/main.c
@@ -10,8 +10,8 @@
 #include <device.h>
 #include <drivers/i2c.h>
 
-#ifdef ARDUINO_I2C_LABEL
-#define I2C_DEV ARDUINO_I2C_LABEL
+#ifdef DT_ALIAS_ARDUINO_I2C_LABEL
+#define I2C_DEV DT_ALIAS_ARDUINO_I2C_LABEL
 #else
 #define I2C_DEV "I2C_0"
 #endif
```

frantony, 2019-11-05:
@pabigot 

Thanks for your explanation!

erwango, 2019-11-05:
@ydamigos, I let you check but according to @pabigot analysis, I don't think this is a bug, but rather a improvement.

pabigot, 2019-11-05:
> I don't think this is a bug, but rather a improvement.

If "this" is the sample not selecting the right I2C bus: I think that is a bug because it broke the sample, but it's in the infrastructure related to identifying Arduino resources in devicetree.  Nothing specific to ST, and I've pulled that out as #20330.

If "this" is the sample not working on ST when the bus is correctly identified: I can see an argument it's not a bug because the sample uses an SMBUS command, not an I2C command, and (perhaps) that just doesn't work on ST.  (I suspect it could be made to work, but that the ST implementation notices the non-address transaction length is zero and doesn't initiate a bus transaction at all.)  That would be an enhancement to the ST I2C implementation; I've updated the issue category.

Thanks for pointing this out.

erwango, 2019-11-05:
@pabigot, thanks for confirming.

ydamigos, 2019-11-10:
@erwango, @pabigot I believe it is a bug because I2C driver didn't even try to send the zero length message.

closed 2019-11-12 by ioannisg

frantony, 2019-11-12:
@ioannisg 

#20560 does not fix the ```I2C: Device driver not found.``` problem.

Please reopen the issue.

pabigot, 2019-11-12:
The "does not work" part has been fixed, which was specific to this issue.  The general issue of #20330 remains open to address the device-not-found error; please follow that to check for solutions.

frantony, 2019-11-13:
The "does not work" part has not been fixed because the erroneous behavior (see in the issue header) is still reproducible.

PR #20560 (2019-11-12): i2c_ll_stm32: Allow to send zero length message
Some applications, like i2c_scanner, require to send zero length messages.

Tested on disco_l475_iot1 with `samples/driversi2c_scanner`.
Tested also `samples/sensor/hts221` and `samples/display/cfb_shell` to 
make sure I didn't break something.

Fixes #20325