zephyrproject-rtos/zephyr #20330: devicetree Arduino bindings do not support identification of bus controllers

pabigot, 2019-11-05:
(split off from #20325)

The commit 442df97366416a2779882d185c94e36619a1223a (scripts/dts: Remove alias defines for labels) breaks the existing use of a node label to identify Arduino bus resources.  The commit was incorrect in stating that the generated label was never used.  There is one in-tree use, in `samples/drivers/i2c_scanner`.

If it is desirable to identify Arduino bus resources, then either a fix like the following must be applied to all Arduino binding overlays, or some new mechanism that generates macros based on node labels (not `label` property) must be added to the scripting.

```
diff --git a/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi b/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
index 1f47915822..c6a37a6ffd 100644
--- a/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
+++ b/boards/arm/nucleo_l476rg/arduino_r3_connector.dtsi
@@ -5,6 +5,9 @@
  */
 
 / {
+       aliases {
+               arduino-i2c = &i2c1;
+       };
        arduino_header: connector {
                compatible = "arduino-header-r3";
                #gpio-cells = <2>;
@@ -34,5 +37,3 @@
                           <21 0 &gpiob 8 0>;   /* D15 */
        };
 };
-
-arduino_i2c: &i2c1 {};
diff --git a/samples/drivers/i2c_scanner/src/main.c b/samples/drivers/i2c_scanner/src/main.c
index 44c1724377..ac71a0d85f 100644
--- a/samples/drivers/i2c_scanner/src/main.c
+++ b/samples/drivers/i2c_scanner/src/main.c
@@ -10,8 +10,8 @@
 #include <device.h>
 #include <drivers/i2c.h>
 
-#ifdef ARDUINO_I2C_LABEL
-#define I2C_DEV ARDUINO_I2C_LABEL
+#ifdef DT_ALIAS_ARDUINO_I2C_LABEL
+#define I2C_DEV DT_ALIAS_ARDUINO_I2C_LABEL
 #else
 #define I2C_DEV "I2C_0"
 #endif
```

ulfalizer, 2019-11-05:
How do I reproduce this failure? The sample builds when I do

```
$ cmake -GNinja -DBOARD=arduino_due ..
$ ninja
```

ulfalizer, 2019-11-05:
Ah... never mind. Didn't notice the last part of the diff.

galak, 2020-02-19:
I think this will be resolved by the NODELABEL support from issue #22555 and PR #22612

pabigot, 2020-04-06:
This now works:
```
diff --git a/samples/drivers/i2c_scanner/src/main.c b/samples/drivers/i2c_scanner/src/main.c
index 9e4d3cc86b..7fd0e47c79 100644
--- a/samples/drivers/i2c_scanner/src/main.c
+++ b/samples/drivers/i2c_scanner/src/main.c
@@ -10,8 +10,8 @@
 #include <device.h>
 #include <drivers/i2c.h>
 
-#ifdef ARDUINO_I2C_LABEL
-#define I2C_DEV ARDUINO_I2C_LABEL
+#if DT_HAS_NODE(DT_NODELABEL(arduino_i2c))
+#define I2C_DEV DT_LABEL(DT_NODELABEL(arduino_i2c))
 #else
 #define I2C_DEV "I2C_0"
 #endif
```

closed 2020-04-06 by pabigot