zephyrproject-rtos/zephyr #20340: Define a consistent naming convention for DTS aliases to be used by Zephyr tests/samples

galak, 2019-11-05:
Currently we have a random mix of device tree aliases we utilize in sample & test code.
[ See #20334 for list of aliases used].

I think we've reasonable come to the point that utilizing aliases for samples/test is a good practice.  However, we haven't utilized a standard convention for these aliases like we have for chosen properties.

This issue is being opened to start the discussion on what naming scheme we should try and utilize.

erwango, 2019-11-05:
Also, in #20334, aliases are defined in .dtsi. I'm wondering if this is the good thing to do.

galak, 2019-11-05:
Initial proposal would be:

`zephyr,test-<DEV>-<INSTANCE>` or `zephyr,test-<DEV>-<NAME>`

So we'd have:
```
zephyr,test-led-0
zephyr,test-led-1
zephyr,test-sw-0
zephyr,test-pwm-led-0
zephyr,test-pwm-led-red /* example of name */
zephyr,test-pwm-led-green /* example of name */
zephyr,test-pwm-led-blue /* example of name */
zephyr,test-pwm-0
zephyr,test-pwm-1
zephyr,test-pwm-2
zephyr,test-pwm-3
zephyr,test-i2c-0
zephyr,test-i2c-1
zephyr,test-i2c-2
zephyr,test-gpio-0
zephyr,test-gpio-1
```

This would get us defines of the form:

```
DT_ALIAS_ZEPHYR_TEST_I2C_0_...
DT_ALIAS_ZEPHYR_TEST_PWM_LED_RED_..
```

galak, 2019-11-05:
> Also, in #20334, aliases are defined in .dtsi. I'm wondering if this is the good thing to do.

Agreed, for aliases utilized for samples/test I think they should be in the board dir only.  Maybe some rare case exists, but in general the board should set the alias if it can use/support the sample/test.

pabigot, 2019-11-05:
I'm not entirely convinced that there's a need for generic aliases like this at the board level, certainly all of the ones listed.  For example, we've already got a standard for `led-0` identifying an LED; why would we need to also have `zephyr,test-led-0`?

And we need to be careful about things like `zephyr,test-gpio-0`.  #20334 would have that be a controller, but that's not useful in a test: we need it to be a pin.  Then we need to clearly document any requirements on a suitable target for the alias.  For example, the GPIO two-pin API test requires that `gpio-0` and `gpio-1` be on the same controller, and not connected to anything that affects their electrical behavior (e.g. pull-ups).  (The test also requires that they be shorted, but that is a test-specific requirement.)

Similarly `zephyr,test-i2c-0` when identifying a bus is useful for only one thing: scanning for available devices.  I'm having difficulty imagining where we'd need more than one of most of those things.

If we do add documented board-level aliases, I would like to have `zephyr,xtest-` documented as reserved for the prefix for aliases that will be used only in specific tests (where the alias appears in a devicetree overlay in the test directory).  We might never use it, but if we need it we won't need to argue about it.

Also if we're doing this we should also document other standard aliases, e.g. `arduino,i2c` or `feather,i2c`.

erwango, 2024-02-13:
@hakehuang I'm assigning this to you as part of your work on tests and samples.
Feel free to close if you feel to.

hakehuang, 2024-02-14:
@erwango , I would like to keep it open but link this to [#68400 ](https://github.com/zephyrproject-rtos/zephyr/issues/68400), we are unify the defines as well

hakehuang, 2024-02-14:
> Initial proposal would be:
> 
> `zephyr,test-<DEV>-<INSTANCE>` or `zephyr,test-<DEV>-<NAME>`

I would propose it to be `zephyr,<test_interface>-<DEV>-<INSTANCE>`   as we are trying to introduce a common framework to host all common interfaces.

such as we will have
```
zephyr,nnboard-led-0
```
and also have
```
zephyr,arduino-led-0
```