zephyrproject-rtos/zephyr #20348: Device Tree Enhancements/Changes/Fixes for 2.2

galak, 2019-11-05:
Catch all issue to track what DT related enhancements / changes / fixes we are aiming for in the Zephyr 2.2 Release.

- [ ] Convert Sensor drivers to DT - #9962

pabigot, 2019-11-28:
@galak Should this include #20253 or at least some milestones toward that?

pabigot, 2019-12-10:
In fact I'd like to see progress on as many of these as possible:

* #19904 better support for one-driver-multiple-devices (compatibles)
* #20287 instance number inconsistency with multiple compatibles
* #19448 support devices enabled/disabled at runtime (requires device API change)
* #19285 which could help resolve #20325
* #20253 which could ensure initialization order that supports dependencies

pabigot, 2020-03-10:
@carlescufi Why did you change the name of this?  I thought the original name was right (and this *is* the issue I considered the roll-up list).

carlescufi, 2020-03-11:
> @carlescufi Why did you change the name of this? I thought the original name was right (and this _is_ the issue I considered the roll-up list).

Because I wanted something descriptive for the Release Plan project. Hence also why I asked @galak to move your comment to wherever it is appropriate.

galak, 2020-04-01:
Posted several PRs for entropy DTS conversion.  After those PRs, left with NRF and STM32 as the outstanding drivers that need conversion.

galak, 2020-04-06:
NRF and STM32 entropy drivers now using DTS.

closed 2020-04-06 by galak