zephyrproject-rtos/zephyr #20351: sample vl053lx fails on disco_l475_iot1

erwango, 2019-11-05:
**Describe the bug**
Sensor init fails.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/sensor/vl53l0x
2. west build -b disco_l475_iot1
3. west flash
4. See error

**Expected behavior**
Sensor init succeeds  and report value

**Impact**
Cant' test sensor

**Screenshots or console output**
```
0x08006520 in LL_I2C_IsActiveFlag_BERR (I2Cx=0x40005800) at /local/mcu/zephyr/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/stm32l4xx_ll_i2c.h:1617
1617	  return ((READ_BIT(I2Cx->ISR, I2C_ISR_BERR) == (I2C_ISR_BERR)) ? 1UL : 0UL);
(gdb) bt
#0  0x08006520 in LL_I2C_IsActiveFlag_BERR (I2Cx=0x40005800) at /local/mcu/zephyr/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/stm32l4xx_ll_i2c.h:1617
#1  check_errors (funcname=<optimized out>, dev=<optimized out>) at /local/mcu/zephyr/zephyr-project/drivers/i2c/i2c_ll_stm32_v2.c:500
#2  0x080065d2 in stm32_i2c_msg_write (dev=dev@entry=0x20001550 <__device_i2c_stm32_I2C_2>, msg=msg@entry=0x200006c4 <z_main_stack+668>, next_msg_flags=next_msg_flags@entry=0x0, slave=slave@entry=41)
    at /local/mcu/zephyr/zephyr-project/drivers/i2c/i2c_ll_stm32_v2.c:555
#3  0x08006708 in i2c_stm32_transfer (dev=0x20001550 <__device_i2c_stm32_I2C_2>, msg=<optimized out>, num_msgs=1 '\001', slave=<optimized out>) at /local/mcu/zephyr/zephyr-project/drivers/i2c/i2c_ll_stm32.c:138
#4  0x080067c4 in z_impl_i2c_transfer (addr=<optimized out>, num_msgs=1 '\001', msgs=0x200006c4 <z_main_stack+668>, dev=<optimized out>) at ../../../../../include/drivers/i2c.h:252
#5  i2c_transfer (addr=<optimized out>, num_msgs=1 '\001', msgs=0x200006c4 <z_main_stack+668>, dev=<optimized out>) at zephyr/include/generated/syscalls/i2c.h:46
#6  i2c_write (dev=<optimized out>, buf=buf@entry=0x200006dc <z_main_stack+692> "", num_bytes=num_bytes@entry=2, addr=<optimized out>) at ../../../../../include/drivers/i2c.h:395
#7  0x0800684e in i2c_reg_write_byte (value=value@entry=0 '\000', reg_addr=reg_addr@entry=0 '\000', dev_addr=<optimized out>, dev=<optimized out>) at ../../../../../include/drivers/i2c.h:568
#8  VL53L0X_WrByte (Dev=Dev@entry=0x20000190 <vl53l0x_driver+4>, index=index@entry=0 '\000', data=data@entry=0 '\000') at /local/mcu/zephyr/zephyr-project/drivers/sensor/vl53l0x/vl53l0x_platform.c:65
#9  0x0800529c in VL53L0X_get_info_from_device (Dev=Dev@entry=0x20000190 <vl53l0x_driver+4>, option=option@entry=2 '\002') at /local/mcu/zephyr/modules/hal/st/sensor/vl53l0x/api/core/src/vl53l0x_api_core.c:246
#10 0x08001dfe in VL53L0X_check_part_used (Dev=0x20000190 <vl53l0x_driver+4>, Revision=0x2000076f <z_main_stack+839> " c", pVL53L0X_DeviceInfo=0x2000078c <z_main_stack+868>)
    at /local/mcu/zephyr/modules/hal/st/sensor/vl53l0x/api/core/src/vl53l0x_api_strings.c:55
#11 0x08001e38 in VL53L0X_get_device_info (Dev=Dev@entry=0x20000190 <vl53l0x_driver+4>, pVL53L0X_DeviceInfo=pVL53L0X_DeviceInfo@entry=0x2000078c <z_main_stack+868>)
    at /local/mcu/zephyr/modules/hal/st/sensor/vl53l0x/api/core/src/vl53l0x_api_strings.c:83
#12 0x080044f6 in VL53L0X_GetDeviceInfo (Dev=Dev@entry=0x20000190 <vl53l0x_driver+4>, pVL53L0X_DeviceInfo=pVL53L0X_DeviceInfo@entry=0x2000078c <z_main_stack+868>)
    at /local/mcu/zephyr/modules/hal/st/sensor/vl53l0x/api/core/src/vl53l0x_api.c:108
#13 0x0800341c in vl53l0x_init (dev=0x20001568 <__device_vl53l0x>) at /local/mcu/zephyr/zephyr-project/drivers/sensor/vl53l0x/vl53l0x.c:242
```

**Environment (please complete the following information):**
 - OS: : Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA: d6d79933c1a991bc4456a9cadd000b2f638e2891

**Additional context**
Was failing on 2.0

FRASTM, 2019-11-13:
I cannot see this error.
Included  i2c_ll_stm32: Allow to send zero length message [#20560]
In the VL53L0X_get_info_from_device, writing  0x00, 0x00 to the i2c, always gives a length of 2 when writing to the sensor.

avisconti, 2019-11-14:
When testing with d6d7993 I do not see any message on console. I'nm not sure is the same error, but seems so.

Starting from f2b75fd on, everything works.

@erwango, can you cross verify this?

erwango, 2019-11-21:
@avisconti, I confirm your observation.
What is strange is that we don't even get banner when test is not working.
Though, I don't see link with f2b75fd644c87c6cc266a61865b55205cc13b8b9.

Anyway, this is working, so I don't see reason for keeping it open..
Closing

closed 2019-11-21 by erwango