zephyrproject-rtos/zephyr #20355: west build for zephyr/samples/net/sockets/echo_server/ on qemu_xtensa target outputs elf with panic

antfobe, 2019-11-05:
**Describe the bug**
I am trying to get zephyr network examples to run under qemu_xtensa target - west outputs the elfs, however when trying to run them on qemu with `west build -t run` they generate the following panic:
`-- west build: running target run                                                                                                                 
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: sample_controller 
                                                                          
[00:00:00.010,000] <err> os:  ** FATAL EXCEPTION                                                                                                  

[00:00:00.010,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)

[00:00:00.010,000] <err> os:  **  PC 0x6000986c VADDR 0x00000004

[00:00:00.010,000] <err> os:  **  PS 0x00060220

[00:00:00.010,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:2 CALLINC:2)

[00:00:00.010,000] <err> os:  **  A0 0xa0019da3  SP 0x6001eff0  A2 0x0000001d  A3 0x6000040c

[00:00:00.010,000] <err> os:  **  A4 0x60027714  A5 0x00000400  A6 0x6001e820  A7 0x00000001

[00:00:00.010,000] <err> os:  **  A8 0x600281b4  A9 0x6001efe0 A10 0x00000000 A11 0x00000000

[00:00:00.010,000] <err> os:  ** A12 0x00060420 A13 0xffffffff A14 0x00000001 A15 0x00000000

[00:00:00.010,000] <err> os:  ** SAR 0x00000000

[00:00:00.010,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception

[00:00:00.010,000] <err> os: Current thread: 0x60028bf8 (unknown)

[00:00:00.010,000] <err> os: Halting system
`

**To Reproduce**
Steps to reproduce the behavior:
1. Have zephyr-sdk and esp-idf installed;
2. build any network example with qemu_xtensa target
3. attempt to run with qemu
4. See panic

**Expected behavior**
It would be expected that the qemu_xtensa elf would run exactly like the qemu_x86 one for these examples.

**Impact**
This bug is a showstopper for my efforts.

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux (uname -a : `Linux std4rch 5.3.7-arch1-2-ARCH #1 SMP PREEMPT @1572002934 x86_64 GNU/Linux`)
 - Toolchain: Zephyr v0.10.3; SDK, ESP-IDF v4.1-dev-592-g758db1e00-dirty
 - Commit SHA: 5881f118c0a4dbef8b6534327a666e4fa156ce65

**Additional context**
Tested hello_world, synchronization and philosophers and they work fine;
My cmds:
`(./tools/net-tools/loop-socat.sh &) && sudo ./tools/net-tools/loop-slip-tap.sh`
`ESP_IDF_PATH=/opt/esp-idf/ ESPRESSIF_TOOLCHAIN_PATH=/opt/zephyr-os/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/ ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-os/zephyr-sdk/ ZEPHYR_TOOLCHAIN_VARIANT=zephyr west build -b qemu_xtensa zephyr/samples/net/sockets/echo_server/ -- -DCONF_FILE="prj.conf"`

My prj.conf:
[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/3810276/prj.conf.txt)
CMake logs:
[CMakeOutput.log](https://github.com/zephyrproject-rtos/zephyr/files/3810333/CMakeOutput.log)
[CMakeError.log](https://github.com/zephyrproject-rtos/zephyr/files/3810334/CMakeError.log)

antfobe, 2019-11-06:
Tried adding the QEMU settings at https://github.com/zephyrproject-rtos/net-tools to my prj.conf:
[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/3816052/prj.conf.txt)

Still getting the panics. Can't really dig into these sdk problems hinted at the CMake error log:
[CMakeError.log](https://github.com/zephyrproject-rtos/zephyr/files/3816068/CMakeError.log)

antfobe, 2019-11-06:
Welp got an error related to the settings `CONFIG_NET_LOG=y` and `CONFIGT_LOG=y`

`/opt/zephyr-os/zproj/zephyr/subsys/logging/log_backend_xtensa_sim.c:31:2: error: invalid hard register usage between output operands
   31 |  __asm__ __volatile__ (
      |  ^~~~~~~`

which probably is responsible for display qemu output to stdout - so still same panic.
- I used `./go.sh xtensa_sample_controller` for sdk-ng and manually downloaded isl from [waybackmachine](https://web.archive.org/web/20180510193902/http://isl.gforge.inria.fr/isl-0.18.tar.xz)

antfobe, 2019-11-13:
@jukkar can I get any update on this?

jukkar, 2019-11-26:
Sorry for late reply, just returned from vacation.

This could be stack size issue, can you try to increase the size of the networking thread stacks, the relevant options are:
CONFIG_NET_TX_STACK_SIZE (default value is 1200)
CONFIG_NET_RX_STACK_SIZE (default value is 1500)

antfobe, 2019-11-26:
Increased them both to 2048, still getting the panic:
[image]

jukkar, 2019-11-27:
> Isn't it related to the network stack? I was getting similar panics in FreeRTOS, until I fixed the esp-idf calls (esp_netif_init or something), the espressif guys changed them in the new release.

So what you are saying above means that the issue is not in network stack but somewhere else.

antfobe, 2019-11-28:
> So what you are saying above means that the issue is not in network stack but somewhere else.

Yes, pretty much - how do you pass the network interfaces (like e1000, openeth) to zephyr?

jukkar, 2019-11-29:
> Yes, pretty much - how do you pass the network interfaces (like e1000, openeth) to zephyr?

First one needs to create zeth interface in host
```
$ cd $ZEPHYR_BASE/../net-tools
$ ./net-setup.sh
```
Then compile zephyr to use the e1000 driver. For this, there is a overlay-e1000.conf file in echo-server sample.
```
$ cd samples/net/sockets/echo_server
$ cmake -Bbuild -DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-e1000.conf .
$ make -C build run
```

jukkar, 2019-11-29:
Unassigning myself as this is not networking issue.

antfobe, 2019-11-29:
> Unassigning myself as this is not networking issue.

As pointed out in the issues' name, its an xtensa/qemu_xtensa bug - the build should work for other targets, not only qemu_x86.

antfobe, 2019-11-29:
Well at least the error changed with `-DOVERLAY_CONFIG=overlay-qemu_802154.conf`:

[image]

antfobe, 2019-12-03:
@jukkar, could you assign this issue to someone that is related to xtensa development in zephyr? - or if you were able to make any network example work, post the steps to reproduce your working example?

jukkar, 2019-12-03:
```
$ grep xtensa CODEOWNERS 
/soc/xtensa/intel_s1000/                  @sathishkuttan @dcpleung
/arch/xtensa/                             @andrewboie @dcpleung @andyross
/soc/xtensa/                              @andrewboie @dcpleung @andyross
/boards/xtensa/                           @nashif @dcpleung
/boards/xtensa/intel_s1000_crb/           @sathishkuttan @dcpleung
/boards/xtensa/odroid_go/                 @ydamigos
/dts/xtensa/xtensa.dtsi                   @ydamigos
/include/arch/xtensa/                     @andrewboie
```
Lets pick dcpleung from the list.

dcpleung, 2019-12-03:
The sample is not whitelist-ed to run on `qemu_xtensa` as indicated in `sample.yaml`. This is probably due to lack of UART driver for `qemu_xtensa`. Can you confirm that the fault instruction address (PC, in your example 0x600092d0) in `zephyr/zephyr.lst` points to `api->poll_in(dev, p_char)` in `z_impl_uart_pool_in()`?

antfobe, 2019-12-03:
Yes, I think this confirms it:
[image]

dcpleung, 2019-12-03:
According to http://wiki.osll.ru/doku.php/etc:users:jcmvbkbc:qemu-target-xtensa, the sim platform does not seem to have serial support (contrary to other platforms having 16550s).

Closing this as this is not a problem of Zephyr but a lack of feature in QEMU.

closed 2019-12-03 by dcpleung

antfobe, 2019-12-03:
Reading the page above, the lx200 machine should support UART, by modifying ninja's call to qemu in `build/build.ninja` to:
`/opt/zephyr-os/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-xtensa -machine lx200 -semihosting -nographic -cpu sample_controller -net none -pidfile qemu.pid -serial mon:stdio -kernel /opt/zephyr-os/zproj/build/zephyr/zephyr.elf`
The build still panicked:
[image]

dcpleung, 2019-12-03:
There is still no driver in Zephyr, and so nothing for the shell to attach. This results in the shell calling UART function with a NULL device pointer.

antfobe, 2019-12-03:
I see - well, at least I managed to get some networking with [espressif's qemu](https://github.com/espressif/qemu/wiki)
Maybe zephyr will have this driver in the future, thanks anyway.

dcpleung, 2019-12-03:
If you can figure out the MMIO address, you can try enabling the existing 16550 driver and see if it works.

jcmvbkbc, 2019-12-04:
UART base address is 0xfd050020 in lx60/lx200/ml605/kc705 machines.

antfobe, 2019-12-05:
Have no idea what I'm doing, trying to follow this [example](https://dangokyo.me/2018/03/27/qemu-internal-memory-region-address-space-and-qemu-io/) on qemu. If I'm able to make it work i'll post my results here.