zephyrproject-rtos/zephyr #20358: Set Kconfig BOARD via env / cmake BOARD variable

galak, 2019-11-05:
Right now we have the possibility of being slightly out of sync between the 'BOARD' Kconfig option and what we specify on the command line to cmake via -DBOARD.  We should have the Kconfig BOARD variable set via being passed in via cmake.

See discussion on PR #20084

erwango, 2019-11-07:
Looking at discussion in PR #20084, wonder how things will end up for stuff like dual core SoCs where we define 2 targets for a same board:

For boards who have 2 different cores:
`boards/arm/stm32h747i_disco/Kconfig.defconfig`:
```
if BOARD_STM32H747I_DISCO_M7 || BOARD_STM32H747I_DISCO_M4

config BOARD
	default "stm32h747i_disco_m7" if BOARD_STM32H747I_DISCO_M7
	default "stm32h747i_disco_m4" if BOARD_STM32H747I_DISCO_M4
...
```

For boards who have different revision:
`boards/arm/stm32_min_dev`
```
if BOARD_STM32_MIN_DEV_BLUE || BOARD_STM32_MIN_DEV_BLACK

config BOARD
	default "stm32_min_dev_blue" if BOARD_STM32_MIN_DEV_BLUE
	default "stm32_min_dev_black" if BOARD_STM32_MIN_DEV_BLACK
...
```

henrikbrixandersen, 2022-11-17:
Related: #8566

nordicjm, 2024-11-19:
Fixed in hwmv2

closed 2024-11-19 by nordicjm