zephyrproject-rtos/zephyr #20388: Allow for runtime reconfiguration of SPI master / slave

raveslave, 2019-11-06:
It would be great if there was a way to do run-time changes between SPI master to slave and back while the system is running.

This can be useful when bootstrapping an FPGA during boot. Once booted yo might want to re-use the same pins for something else, i.e. SPI slave, UART, etc.

carlescufi, 2019-11-07:
@tbursztyka is currently working on enhancing the SPI API, so copying him here in case this can be taken into account.

tbursztyka, 2020-01-09:
But that's already possible. There is no limit in how many spi_config you can use on a controller.

If a controller supports both being master and a slave, that's fine. Just have one spi_config with SPI_OP_MODE_MASTER. And then use another spi_config with SPI_OP_MODE_MASK.

 The hardware being the limit obviously here.

When it comes to pins however, that has nothing to do with SPI API at all. It's pinmuxing etc.. I believe that's already possible to do, it just requires you to do it at relevant places in your code.

closed 2020-01-09 by tbursztyka

raveslave, 2020-01-13:
@tbursztyka but can you have both SPI_OP_MODE_MASTER & SPI_OP_MODE_MASK for the same physical SPI, allowing to switch between the two runtime?