zephyrproject-rtos/zephyr #20392: Bluetooth stack degrading timing of cooperative threads

wopu-ot, 2019-11-06:
**Describe the bug**

The Bluetooth stack uses the system workqueue to execute "heavy" encryption functions. At the same time the Bluetooth subsystem insists on running the system workqueue as cooperative thread. This degrades the timing of all cooperative threads in the system.

**To Reproduce**

Run Bluetooth stack and trigger encryption. The function `bt_smp_aes_cmac` will be called from the system workqueue, e.g. through `z_work_q_main`->`l2cap_rx_process`->`l2cap_chan_recv`->`bt_smp_recv`->`smp_dhkey_check`->`compute_and_check_and_send_slave_dhcheck`->`smp_f5`. No other cooperative thread can execute while encryption functions are executed for several milliseconds, thus introducing considerable jitter.

**Expected behavior**

All cooperative threads behave "nicely" and do not hog the CPU.

**Impact**

Threads handling audio data may be affected to a point where audio quality is degraded.

**Additional context**

- I tried running the system workqueue as preemptive thread and nothing bad seemed to happen. Is anything *known* to break when running the system workqueue as preemptive thread?
- Using Meta-IRQs for more time-sensitive processing is not an option, as these are broken themselves #20255

jhedberg, 2019-11-06:
I wonder if we could somehow repurpose ecc_thread in hci_ecc.c for this. It has a pre-emptible priority. So far it's been used to emulate the standard ECC HCI commands.

carlescufi, 2019-11-14:
>I tried running the system workqueue as preemptive thread and nothing bad seemed to happen. Is anything known to break when running the system workqueue as preemptive thread?

I am quite sure there are several places in the code, run by the system workqueue, where we assume non-preemptability. There is a long-term goal of making all Bluetooth threads preemptible, but that requires more careful analysis.

carlescufi, 2019-11-14:
>Using Meta-IRQs for more time-sensitive processing is not an option, as these are broken themselves #20255

There is a PR fixing those, so that could be a short-term solution. Converting the current ecc thread to handle all SMP crypto-related calls could be a very good option as well.

wopu-ot, 2019-11-15:
Fixing #20255 makes it somewhat less urgent to fix this issue and I would be okay with lowering its priority.

aescolar, 2019-11-15:
Reclassifying as an enhancement