zephyrproject-rtos/zephyr #20406: misc.app_dev.libcxx test fails to build for qemu_x86_64

stephanosio, 2019-11-07:
**Describe the bug**
`misc.app_dev.libcxx` test fails to build for `qemu_x86_64` due to `-lstdc++` not being specified while linking.

Since `misc.app_dev.libcxx` is run as part of `sanitycheck` for x86 arch, `sanitycheck` also fails. This seems to have gone unnoticed because CI does not currently build and run this test on `qemu_x86_64`.

_Unrelated note: x86_64 is classified as x86 for some reason for both Kconfig ARCH and testcase arch; this should be addressed and fixed separately (maybe x86 should also follow how riscv arch handles riscv32 and riscv64)._

**To Reproduce**
Steps to reproduce the behavior:
1. `cd tests/application_development/libcxx`
2. `mkdir build && cd build`
3. `cmake -DBOARD=qemu_x86_64 ..`
4. `make`

**Screenshots or console output**
Build fails with the following error:
```
FAILED: zephyr/zephyr.elf zephyr/zephyr.bin zephyr/zephyr.lst zephyr/zephyr.stat
: && ccache /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++  -m64   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr.elf  -Wl,-T zephyr/linker.cmd -Wl,-Map=/home/stephanos/Dev/zephyr-public/zephyr/tests/application_development/libcxx/build_2/zephyr/zephyr_prebuilt.map -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/arch/x86/core/libarch__x86__core.a zephyr/lib/libc/newlib/liblib__libc__newlib.a zephyr/lib/posix/liblib__posix.a zephyr/soc/x86/ia32/libsoc__x86__ia32.a zephyr/subsys/random/libsubsys__random.a zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -L/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/lib64 -L"/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/64" -L/home/stephanos/Dev/zephyr-public/zephyr/tests/application_development/libcxx/build_2/zephyr -lgcc -Wl,--print-memory-usage -m64 -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -u_OffsetAbsSyms -u_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn -lm -lc -lgcc -L"/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf"/lib/64 && cd /home/stephanos/Dev/zephyr-public/zephyr/tests/application_development/libcxx/build_2/zephyr && cmake -E rename zephyr_prebuilt.map zephyr.map && /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objcopy -S --gap-fill 0xff --output-target=binary --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.bin && /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objdump -S zephyr.elf > zephyr.lst && /opt/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-readelf -e zephyr.elf > zephyr.stat
Memory region         Used Size  Region Size  %age Used
             RAM:       87088 B         4 MB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `__gnu_cxx::new_allocator<int>::deallocate(int*, unsigned long)':
/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/ext/new_allocator.h:125: undefined reference to `operator delete(void*)'
/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `std::_MakeUniq<make_unique_data>::__single_object std::make_unique<make_unique_data>()':
/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/bits/unique_ptr.h:831: undefined reference to `operator new(unsigned long)'
/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `std::default_delete<make_unique_data>::operator()(make_unique_data*) const':
/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/bits/unique_ptr.h:81: undefined reference to `operator delete(void*, unsigned long)'
/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `__gnu_cxx::new_allocator<int>::deallocate(int*, unsigned long)':
/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/ext/new_allocator.h:125: undefined reference to `operator delete(void*)'
/opt/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/8.3.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `__gnu_cxx::new_allocator<int>::allocate(unsigned long, void const*)':
/opt/zephyr-sdk/x86_64-zephyr-elf/x86_64-zephyr-elf/include/c++/8.3.0/ext/new_allocator.h:111: undefined reference to `operator new(unsigned long)'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
```

In terms of `sanitycheck -v -N -l -p qemu_x86_64`, refer to the following link for full log:
https://gist.github.com/stephanosio/d8ac379bced247b7f353a00382e9ebb1

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.3
 - Commit: 7e722564df1586b722314146c8a427e48249f536

**Additional context**
This seems to have been broken for a while, at least for more than a week.

It seems to be more of a CMake problem actually:
https://github.com/zephyrproject-rtos/zephyr/blob/7e722564df1586b722314146c8a427e48249f536/cmake/extensions.cmake#L1350

Changing this to `zephyr_check_compiler_flag(C ${option} ${check})` does add `-lstdc++` back and the test successfully builds.

stephanosio, 2019-11-07:
cc @andrewboie @SebastianBoe

SebastianBoe, 2019-11-07:
> due to -lstdc++ not being specified while linking

I see -lstdc++ has been added through "zephyr_ld_options" and not "zephyr_link_libraries". ld_options will test whether the flag is compatible with the toolchain and silently omit it if not, whereas link_libraries just applies it.

If the flag was not optional then it should have been using the zephyr_link_libraries API.

This still begs the question of why the flag failed the compatiblity check. But we could leave this as an open question and just change from zephyr_ld_options to zephyr_link_libraries.

stephanosio, 2019-11-07:
> If the flag was not optional then it should have been using the zephyr_link_libraries API.

I created a PR addressing this: #20415.

> This still begs the question of why the flag failed the compatiblity check. But we could leave this as an open question and just change from zephyr_ld_options to zephyr_link_libraries.

I have seen `linker_ld_options` incorrectly filter out a valid linker option on multiple occasions. I think this deserves to be further investigated.

closed 2019-11-07 by stephanosio