zephyrproject-rtos/zephyr #20411: samples: lis3mdl trigger not working with x_nucleo_iks01a1

erwango, 2019-11-07:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Following, https://github.com/zephyrproject-rtos/zephyr/pull/20343 enables defining irq pin with dts for lis3mdl sensor.
samples/shields/x_nucleo_iks01a1 was updated to exercise lis3mdl trigger but it is not working.
Since there was no samples to use it before, it might not be a regression.

**Impact**
Can use triggering mode for this sensor

**Screenshots or console output**
```
HTS221: Temperature: 25.9 C
HTS221: Relative Humidity: 50.0%
LPS25HB: Pressure:99.0 kpa
LIS3MDL: Magnetic field (gauss): x: 0.1, y: -0.0, z: -0.0
LSM6DS0: Acceleration (m.s-2): x: -0.1, y: 0.0, z: 9.7
3:: lis3mdl trig 0
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: a0fbd4336e7e94b1569406f2b48600d10632eed6

**Additional context**
Add any other context about the problem here.

avisconti, 2020-02-18:
I debugged the issue and I understood it. The issue is related to 2 things:

1. In the driver there is a confusion between DRDY (data is ready) and INT (magn data exceeds a predefined threshold), and the sensor has two separate pins for those interrupts. The shield overlay file is mapping the wrong one (INT).

2. When setting the trigger it is needed to perform a dummy read in order to re-trigger interrupts.

So, I am now able to provide a simple fix.

As a general consideration the driver looks old and should be aligned to the more recent ones, even if the h/w sensor itself is old and I'm not sure it is worth the hassle. So, I'll postpone the decision to do the alignment or not. For the moment I'm more on the NO side.

avisconti, 2020-02-18:
Side note:

I noted that currently three boards are using lis3mdl driver: x_nucleo_iks01a1, 96b_stm32_sensor_mez, disco_l475_iot1.

But only x_nucleo_iks01a1 is using irq-gpios, so there is no DRDY vs INT confusion on the other two boards.

PR #22896 (2020-02-25): Fix lis3mdl issue 20411
Fixes #20411

The issue is related to 2 things:

1. In the driver there is a confusion between DRDY (data is ready) and INT (magn data exceeds a predefined threshold), and the sensor has two separate pins for those interrupts. The shield overlay file is mapping the wrong one (INT).

2. When setting the trigger it is needed to perform a dummy read in order to re-trigger interrupts.

EDIT:
Tested on x_nucleo_iks01a1 shield over Nucleo F401RE
closed 2020-02-25 by ioannisg