zephyrproject-rtos/zephyr #20414: sample: bluetooth: mesh_demo: issues (BBC Micro:bit)

WilliamGFish, 2019-11-07:
**Describe the bug**
After compiling the Mesh_Demo sample for the BBC Micro:bit it works  but has issues:
- Intermittently reboots
- Often 'crashes' on booting
- Fatal error reports for ticker_users memory locations

Often the application will run for a few minutes after booting and then either begin rebooting or return a fatal error.

**To Reproduce**
Follow steps indicated described sample, which need to be updated to use the WEST tool.

**Expected behavior**
The sample in 1.14 ran smoothly without issue

**Impact**
Issues with the Micro:bit are always both annoyance and potential showstopper as they are used as demo/training devised by many.

**Screenshots or console output**
**FATAL ERROR**
> [0m[00:00:32.221,679] [1B][1;31m<err> os: Faulting instruction address (r15/pc): 0x20000cdc[1B][0m
> [00:00:32.221,740] [1B][1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[1B][0m
> [00:00:32.221,801] [1B][1;31m<err> os: Current thread: 0x20000f90 (unknown)[1B][0m
> [00:00:32.259,094] [1B][1;31m<err> os: Halting system[1B][0m
> 
[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

INTERMITTENT REBOOT
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> 1 hops
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> 1 hops
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> 1 hops
> 1 hops
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> 1 hops
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Initializing...
> Unicast address: 0x0b0c
> Bluetooth initialized
> Mesh initialized
> Loading stored settings
> Using stored settings
> Subscribing to heartbeat messages
> [00:00:00.114,624] [1B][0m<dbg> bt_hci_core.bt_hci_cmd_send_sync: opcode 0x200a status 0x00[1B][0m
> [1B][1;31m--- 504 messages dropped ---
> [1B][0m[00:00:00.129,608] [1B][0m<dbg> bt_mesh_adv.bt_mesh_scan_cb: len 25: 182a37b1aaf678f238803f1b1911fa7~[1B][0m
> [1B][1;31m--- 13 messages dropped ---
> [1B][0m[00:00:00.129,669] [1B][0m<dbg> bt_mesh_net.bt_mesh_net_recv: rssi -58 net_if 0[1B][0m
> [00:00:00.129,821] [1B][0m<dbg> bt_ctlr_hci_driver.recv_thread: blocking[1B][0m
> [1B][1;31m--- 260 messages dropped ---
> 

**Additional context**
I have attempted to adjust the memory settings in order to oleviate issues but have not been successful.

jhedberg, 2019-11-07:
The sample itself has not been touched for a long time, so these issues must be coming from changes elsewhere in the system. It might be related to the controller, or perhaps to other parts of the system consuming more memory (the micro:bit's thread stacks are very optimized, i.e. they can easily overflow if something starts consuming more memory).

WilliamGFish, 2019-11-08:
I've had a good look at all the new options and still haven't found a set that provides stability.

I think its time to buy some Thingys (52s and 91s)

It would be nice if someone found the settings necessary to get the M:Bits working again.

carlescufi, 2019-11-14:
@WilliamGFish could you please try the sample in current upstream master but enabling
`CONFIG_BT_LL_SW_LEGACY=y`? I suspect this has a lot to do with the split controller now being default.

WilliamGFish, 2019-11-14:
Carles,Great that you know the product as well as you do. Made the change and is working great. I'll put together a PR to fix up the sample. Billy..

carlescufi, 2019-11-15:
@WilliamGFish feel free to send the PR, but please do not close this issue. We will use it to remind us that there are issues with the split link layer and nRF51.

WilliamGFish, 2019-11-15:
> @WilliamGFish feel free to send the PR, but please do not close this issue. We will use it to remind us that there are issues with the split link layer and nRF51.

Shall do...

PR #21260 (2019-12-10): samples: Bluetooth: mesh_demo: Increase ISR_STACK_SIZE value
Split Link Layer implementation uses 80 bytes more ISR stack
in comparison to Legacy Link Layer, hence increase the
required ISR_STACK_SIZE for the BBB micro:bit mesh_demo
sample.

Fixes #20414.
commit 0b878020 (2019-12-10): samples: Bluetooth: Increase ISR_STACK_SIZE value

Split Link Layer implementation uses 80 bytes more ISR stack
in comparison to Legacy Link Layer, hence increase the
required ISR_STACK_SIZE for the BBC micro:bit and other
nRF51 QFAA SoC based mesh and mesh_demo samples.

Fixes #20414.
closed 2019-12-10 by jhedberg