zephyrproject-rtos/zephyr #20416: sample: sensor: fxos8700 issues

WilliamGFish, 2019-11-07:
**Describe the bug**
After compiling the fxos8700 sample for the BBC Micro:bit  fails to boot with a fatal error.

The issue appears to be connected to an update to the nRF series die temperature sensing. (drivers: sensor: temp_nrf5: Code cleanup #17249). The 'blocking' aspects seem to create a interrupt deadlock.

I have not reaised a PR with changes as this has links to the nRF Clock Control Calibration;
```
config CLOCK_CONTROL_NRF_CALIBRATION_TEMP_DIFF
    int "Temperature change triggering calibration (in 0.25 degree units)"
    default 2
    depends on CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION && CLOCK_CONTROL_NRF && !CLOCK_CONTROL_NRF_FORCE_ALT && CLOCK_CONTROL
    help
      Calibration is triggered if the temperature has changed by at least
      this amount since the last calibration.
```

The issue may only exist on the nrf51 series as am unable to test on other devices.

**To Reproduce**
Follow steps indicated described sample, which need to be updated to use the WEST tool.

**Expected behavior**
Works as expected as in v1.14

**Impact**
Showstopper as appears to effect multiple sensors

**Screenshots or console output**

> [00:00:00.000,976] [1B][1;31m<err> os:  xpsr:  0x61000000[1B][0m
> [1B][1;31m--- 3 messages dropped ---
> [1B][0m[00:00:00.001,037] [1B][1;31m<err> os: Faulting instruction address (r15/pc): 0x0000449a[1B][0m
> [00:00:00.001,098] [1B][1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[1B][0m
> [00:00:00.001,159] [1B][1;31m<err> os: Current thread: 0x20000774 (unknown)[1B][0m
> [00:00:00.040,344] [1B][1;31m<err> os: Halting system[1B][0m

**Additional context**
If i amend the temp_nrf.c code to remove the block, the application works (well it boots and allows the accelerometer to be read).

Original Code:
```
	r = clock_control_async_on(data->hfclk_dev, NULL, &clk_data);
	__ASSERT_NO_MSG(!r);

	k_sem_take(&data->device_sync_sem, K_FOREVER);

	r = clock_control_off(data->hfclk_dev, 0);
	__ASSERT_NO_MSG(!r);
```

Amended Code:
```
	nrf_temp_task_trigger(NRF_TEMP, NRF_TEMP_TASK_START);

	k_sem_take(&data->device_sync_sem, K_MSEC(10));
```
Output post changes:

> ***** Booting Zephyr OS build zephyr-v2.0.0-1806-geaa6058ebdcf *****
> Setting sampling frequency
> [00:00:00.014,221] [1B][0m<dbg> FXOS8700.fxos8700_set_odr: Set ODR to 0x30[1B][0m
> [1B][1;31m--- 3 messages dropped ---
> [1B][0m[00:00:00.014,617] [1B][0m<dbg> FXOS8700.fxos8700_set_mt_ths: Set FF_MT_THS to 17[1B][0m
> [00:00:00.016,082] [1B][0m<dbg> temp_nrf5.temp_nrf5_sample_fetch: sample: 104[1B][0m
> [00:00:00.016,174] [1B][0m<dbg> temp_nrf5.temp_nrf5_channel_get: Temperature:26,0[1B][0m
> trigger_handler: sensor_sample_fetch 
> AX=  3.677493 AY= -2.911350 AZ= 12.411541 
> trigger_handler: sensor_sample_fetch 
> AX= 14.097059 AY=  0.612915 AZ=  0.153228 
> trigger_handler: sensor_sample_fetch 
> AX= 16.395492 AY=-14.250289 AZ= -1.072603 
> [00:00:12.276,245] [1B][0m<dbg> FXOS8700.fxos8700_handle_motion_int: FF_MT_SRC 0xa4[1B][0m
> [00:00:12.748,474] [1B][0m<dbg> FXOS8700.fxos8700_handle_motion_int: FF_MT_SRC 0x82[1B][0m
> [00:00:13.063,293] [1B][0m<dbg> FXOS8700.fxos8700_handle_motion_int: FF_MT_SRC 0x9e[1B][0m

pabigot, 2019-11-12:
The problem is with initialization order.  Both the FXOX8700 and the nrf5/temp_nrf5 driver are initialized at `CONFIG_SENSOR_INIT_PRIORITY` in that order.  The FXOS8700 driver initialization suspends while waiting on a semaphore for an I2C operation to complete, allowing the `measure_temperature()` function in `nrf_clock_calibration.c` to be invoked through the work queue before `temp_nrf5_init` function has initialized the temperature sensor.

pabigot, 2019-11-12:
Well, no, that was **a** problem.  [This site](https://tech.microbit.org/hardware/i2c/) shows three variants of the microbit; the FXOS8700 is on variant 2, while mine is a variant 1 with an LSM303AGR.  So I can't confirm that the accelerometer works as the Zephyr board description is for an entirely different device.

Nonetheless, #20641 should fix it for people with variant 2 boards.

PR #20641 (2019-11-28): drivers: nrf_clock_calibration: wait for sensor initialization
When using the RC clock source a periodic calibration is invoked that
involves reading from the die temperature sensor.  The code did not
protect against execution order that caused the periodic calibration
to be invoked before the temperature sensor was initialized.

Update the temperature sensor to detect that it has not been
initialized and so should reject attempts to fetch a reading.

Update the calibration code to do nothing when temperature reading
fails.

Fixes #20416
closed 2019-11-28 by nashif