zephyrproject-rtos/zephyr #20417: BME280 wrong pressure unit?

aqua-developer, 2019-11-07:
**Describe the bug**
When I read the pressure value I get something which seems to be off witha factor 10. The manipulation of the value in the driver code also seems to be mistaken by a factor 10. [Here](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/sensor/bme280/bme280.c#L215).

What have you tried to diagnose or workaround this issue?

**To Reproduce**
The zephyr project [readme ](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/sensor/bme280) sample for the BME280 shows a similar example value. One that is out of range for the sensor. A value that is also physically unreasonable.

The comment in the code says hPa, but then it divides by 1000 which makes me think it's actually reporting mPa.

**Expected behavior**
 1013 hPa is nominal sea level pressure. 101.3 hPa has never ever happened at sea level.  

**Impact**
showstopper when I don't know about it. Annoyance now that I have to compensate my data in hindsight.

pabigot, 2019-11-07:
For reasons unknown to me `SENSOR_CHAN_PRESS` is documented to return a value in kPa.  That appears to be what the code is doing regardless of what the comment says.

"If the code and the comments disagree, both are probably wrong."

aurel32, 2019-11-07:
The sensor subsystem returns the pressure in kPA, so the returned value is actually correct. See:
https://docs.zephyrproject.org/latest/reference/peripherals/sensor.html#_CPPv3N16sensor_interface17SENSOR_CHAN_PRESSE

carlescufi, 2019-11-07:
Closing this as per @aurel32 clarification. Please reopen if still in doubt @aqua-developer.

closed 2019-11-07 by carlescufi

aqua-developer, 2019-11-08:
Fair enough, the code seems to do what it was intended to do then. Thank you. I would however still argue that the intention is confusing since all specifications in the datasheet etc are given in hPa, which is the common unit of measurement for barometric pressure. But maybe that's a different topic.

aurel32, 2019-11-08:
> Thank you. I would however still argue that the intention is confusing since all specifications in the datasheet etc are given in hPa, which is the common unit of measurement for barometric pressure.

The idea of the sensor interface is that it returns a value with the same unit independently of what sensor uses. That way it is possible to replace a sensor by another model with minimal changes to the code. That said I agree that the choice of kPa is indeed a bit weird.