zephyrproject-rtos/zephyr #20418: CONFIG_HEAP_MEM_POOL_SIZE should not be limited

zzzh, 2019-11-07:
**Is your feature request related to a problem? Please describe.**
I need to allocate bigger RAM saze than the limit ( Supported values are: 256, 1024, 4096, and 16384 )

**Describe the solution you'd like**
The size limit should come from hardware, not a config

**Describe alternatives you've considered**
Don't limit the CONFIG_HEAP_MEM_POOL_SIZE anymore

**Additional context**

andrewboie, 2019-11-07:
I think this is a documentation issue, k_mem_pool works with larger heaps than this.

zzzh, 2019-11-08:
@andrewboie 
Thanks, do you know how big it can be?

dleach02, 2020-07-23:
Looking at the underlying implementation, there is no limit to the size of the pool except how much memory is actually available in the system. It isn't obvious why the documentation for the HEAP_MEM_POOL_SIZE even shows various individual sizes since the code is going to eventually define an aligned _mpool_buf_<name> character array. I'll post a PR to change the text in this kconfig item.

PR #27104 (2020-07-24): kernel: Remove supported size comments from HEAP_MEM_POOL_SIZE
The mempool implementation doesn't require specific sizes and can support
arbitrary sizes up to the limit of available memory. The Kconfig documentation
on this configuration was confusing users.

Fixes #20418
commit db36b146 (2020-07-24): kernel: Remove supported size comments from HEAP_MEM_POOL_SIZE

The mempool implementation doesn't require specific sizes and can
support arbitrary sizes up to the limit of available memory. The
Kconfig documentation on this configuration was confusing user.

Fixes #20418
closed 2020-07-24 by ioannisg