zephyrproject-rtos/zephyr #20422: Device with bonds should not accept new keys without user awareness

pdunaj, 2019-11-07:
**Is your enhancement proposal related to a problem? Please describe.**
We have a device A that is bonded with dongle B. Now device C comes into play. C has no bonds and it is replicating the address of B.
A looks for B - it will advertise. Since C is using B's address whitelist will allow the connection as address matches. At the moment Zephyr will succeed the security even though there are no bonds on master. New keys will be generated and new bond will be created. A will not know that its is connected to a wrong device.

User should in general see that something is wrong as C will not be able to connect to B (dongle) and so there will be no data coming from A to the host. But in some cases this can be enough to do some damage. E.g. when user starts to enter system login credentials without checking that device sends data properly.

**Describe the solution you'd like**
I would like for the new key pair generation to be disabled if central has no keys but peripheral has them (for this central). Since the current behavior may be needed for some cases it would be great if there is a switch in Kconfig.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

pdunaj, 2019-11-07:
@joerchan

joerchan, 2019-11-07:
@pdunaj Unauthenticated pairing really shouldn't be allowed in this case. This wouldn't be a problem if the first pairing procedure had been authenticated. Since we wouldn't allow an unauthenticated pairing to replace an authenticated bond.

I have added the option you describe. @jhedberg Should we maybe enable this option by default, or my always do it?

PR #20427 (2019-11-08): Bluetooth: SMP: Add option to disallow unauthenticated re-pairing
Add option to disallow all unauthenticated pairing attempts made by the
peer where an unauthenticated bond already exists.
This would prevent cases where an attacker could copy the peer device
address to connect and start an unauthenticated pairing procedure
to replace the existing bond. To recover the bond information in this
case the bond information has to be explicitly deleted with bt_unpair.

Fixes: #20422
closed 2019-11-08 by jhedberg