zephyrproject-rtos/zephyr #20425: storage/flash_map: flash_area_get_sectors cant fetch sector on devices with non-zero flash base addres

nvlsianpu, 2019-11-07:
**Describe the bug**
The function is implemented in silent assumption that addresses begins form zero.
Flash_map description structures provides partitioning info in absolute notation.

**Origin of the bug:**
The flash API  doesn't allow to fetch information about the flash begin offset.

**solution**
Intorduce API for fetch flash base address and use it for fixing flash_map module.

**related issue**
https://github.com/zephyrproject-rtos/zephyr/issues/20423

nvlsianpu, 2019-11-07:
cc @ioannisg @carlescufi

Laczen, 2019-11-08:
@nvlsianpu, would it not be better to hide the flash base address inside the flash api? So for all flash_read, writes, erase offset 0 refers to the flash base address.

nvlsianpu, 2019-11-08:
After revving STM drivers I see that the uses relative addressing in flash API implementation.
The same for DT partitions description.

If flash driver and flash_map should use relative addressing then this is not a bug.

nvlsianpu, 2019-11-08:
@galak Have you opinion on that?

nvlsianpu, 2019-11-14:
> Flash_map description structures provides partitioning info in absolute notation.

Looks like i recognize this unproperly. Flash_map description uses relative offsets and flash driver API as well.

nvlsianpu, 2019-11-14:
This is NOT an issue then, so I'm closing it.

closed 2019-11-14 by nvlsianpu