zephyrproject-rtos/zephyr #20430: User definable logger tag

koffes, 2019-11-07:
Sometimes, one desires to investigate real-time logs from multiple devices simultaneously.
One can of course print the name, ID or state of a device, but if you have lots of log output, this involves scrolling back in the log to see.

Another scenario is that there is some basic "state" in your device which would like to know at all times. 

A great feature would be to be able to prepend all log messages with a user given string.
A standard log may look like this: 
[00:21:43.946,258] <err> ble did not connect.
Where a user may now get:
`[master][00:21:43.946,258] <err> Did not connect`
or
`[00:21:43.946,258][slave] <err> Did not connect`
 
This would be supported by an API call to the logger that lets you set this string runtime. Each time an err, wrn, inf or dbg is printed, this user defined message is appended.

koffes, 2021-12-07:
Any plans to include such a feature?

nordic-krch, 2021-12-22:
@koffes just opened #41403.

PR #41403 (2022-01-04): logging: Add option to have tag which is prepended to all messages
Tag can be changed at runtime. Feature is enabled by setting maximum tag length to positive value. Additionally, default tag can be configured in Kconfig.

Fixes #20430.
closed 2022-01-04 by MaureenHelm