zephyrproject-rtos/zephyr #20444: sanitycheck error with tests/arch/x86/info.

chen-png, 2019-11-08:
**Describe the bug**
when running tests/arch/x86/info on up squared board using sanitycheck, this test case is actually successful, but the sanitycheck will judge it as a failed test case.

This test case is treated as a harness test, but the output information didn't match any passed case of sanitycheck, so it's judged failed by sanitycheck.

**Screenshots or console output**
**** Booting Zephyr OS build zephyr-v2.0.0-1818-g0b27cacabdab *****

info: the Zephyr x86 platform information tool

MULTIBOOT: boot info structure available.
Flags = 0x00000265
Basic memory map: lower = 252K, upper = 261120K.
Extended memory map was at 0x0001008c (672 bytes).
Framebuffer data not present.

MEMORY MAP: source is Multiboot extended map, max 64 entries
0000000000000000 -> 000000000003efff RAM (252K)
000000000003f000 -> 000000000003ffff UNKNOWN (4K)
0000000000040000 -> 000000000009dfff RAM (376K)
000000000009e000 -> 00000000000fffff UNKNOWN (392K)
0000000000100000 -> 000000000fffffff RAM (261120K)
0000000010000000 -> 0000000012150fff UNKNOWN (34116K)
0000000012151000 -> 0000000077b0cfff RAM (1664752K)
0000000077b0d000 -> 0000000079c2afff UNKNOWN (33912K)
0000000079c2b000 -> 0000000079c43fff ACPI (100K)
0000000079c44000 -> 0000000079ca3fff NVS (384K)
0000000079ca4000 -> 000000007a040fff UNKNOWN (3700K)
000000007a041000 -> 000000007a08cfff UNKNOWN (304K)
000000007a08d000 -> 000000007a3fafff RAM (3512K)
000000007a3fb000 -> 000000007a3fbfff NVS (4K)
000000007a3fc000 -> 000000007a425fff UNKNOWN (168K)
000000007a426000 -> 000000007a964fff RAM (5372K)
000000007a965000 -> 000000007a966fff UNKNOWN (8K)
000000007a967000 -> 000000007affffff RAM (6756K)
000000007b000000 -> 000000007fffffff UNKNOWN (81920K)
00000000d0000000 -> 00000000d0ffffff UNKNOWN (16384K)
00000000e0000000 -> 00000000efffffff UNKNOWN (262144K)
00000000fe042000 -> 00000000fe044fff UNKNOWN (12K)
00000000fe900000 -> 00000000fe902fff UNKNOWN (12K)
00000000fec00000 -> 00000000fec00fff UNKNOWN (4K)
00000000fed01000 -> 00000000fed01fff UNKNOWN (4K)
00000000fee00000 -> 00000000fee00fff UNKNOWN (4K)
00000000ff000000 -> 00000000ffffffff UNKNOWN (16384K)
0000000100000000 -> 000000017fffffff RAM (2097152K)

Kernel exclusions:
0x00100000 -> 0x00103850 (rom)
0x00103860 -> 0x00105664 (ram)
ACPI: 4 CPUs found
CPU #0: APIC ID 0x00
CPU #1: APIC ID 0x02
CPU #2: APIC ID 0x04
CPU #3: APIC ID 0x06

TIMER: new local APIC, configured frequency = 19200000Hz
Using CMOS RTC as reference clock:
start = 363594425, end = 363609681, 15256 cycles
start = 382796905, end = 382810458, 13553 cycles
start = 401996851, end = 402012133, 15282 cycles
start = 421198501, end = 421212063, 13562 cycles
start = 440398858, end = 440414097, 15239 cycles
start = 459600154, end = 459615433, 15279 cycles
start = 478801695, end = 478815246, 13551 cycles
start = 498003648, end = 498017213, 13565 cycles
start = 517204059, end = 517217614, 13555 cycles
start = 536405802, end = 536419359, 13557 cycles
average = 14239Hz

info: complete

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: 0b27cacab

andrewboie, 2019-11-12:
I think we need this test to report success/failure.
If that's not possible, perhaps move it to samples/ and greatly relax the regexes used to determine whether it ran or not.

nashif, 2019-11-12:
the test does not have any regex, so this is the bug. Need to add console harness and capture results. @chen-png can you do this please?

andrewboie, 2019-11-12:
> the test does not have any regex, so this is the bug. Need to add console harness and capture results. @chen-png can you do this please?

Should we move this to samples?

nashif, 2019-11-12:
> Should we move this to samples?

it would be a very useless sample...

andrewboie, 2019-11-12:
> it would be a very useless sample...

It's not very useful as a test either if it is unable to report its own success or failure.

chen-png, 2019-11-13:
> the test does not have any regex, so this is the bug. Need to add console harness and capture results. @chen-png can you do this please?

ok, i will.

PR #20684 (2019-11-14): tests/arch/x86/info: add console harness and regex
use the console handler to handle this test, parse
the output and capture the regex to pass it.

Fixes: #20444
closed 2019-11-14 by nashif