zephyrproject-rtos/zephyr #20445: tests/kernel/critical failed on mimxrt1050_evk board.

chen-png, 2019-11-08:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=mimxrt1050_evk  tests/kernel/critical
2. west flash
3. See error

**Screenshots or console output**
***** Booting Zephyr OS build zephyr-v2.0.0-1818-g0b27cacabdab *****

Running test suite kernel_critical
starting test - test_critical

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.3
 - Commit ID: 0b27cacabd

maksimmasalski, 2019-11-11:
@chen-png Do you think that it is again wrong setup of the watchdog like it was with sam_e70_xplained?

chen-png, 2019-11-11:
> @chen-png Do you think that it is again wrong setup of the watchdog like it was with sam_e70_xplained?

you can disable the watchdog config and have a try.

pabigot, 2019-11-11:
FWIW I can reproduce this on mimxrt1060 and it looks like a stack issue.  I'll keep going and transfer to @agansari if I can't make progress.

pabigot, 2019-11-11:
Confirmed, diagnosed, potential solutions discussed in #20585.  An extension to #15487 described in that issue would solve the problem for this platform, but the failure may apply to multiple targets.

PR #15487 (2019-11-26): Drivers: ARM: SysTick: investigate errors and apply fixes
Several fixes/improvements in the SysTick timer driver, including
- correct calculation of elapsed cycles in the ISR
- avoiding starving the ISR if timeouts gets re-scheduled all the time
- reworking the way elapsed cycles are calculated

Added some documentation in functions and variables, too.

The fixes are closing several ARM timer-related bugs:

Fixes #18591 which is an instance of #20585
Fixes #20445 which is also an instance of #20585 
Also incorporates the SysTick modification related to #20939

### Mandatory Testing in Hardware [... cut]
closed 2019-11-26 by dleach02