zephyrproject-rtos/zephyr #20450: Bluetooth: hci_uart: conn param update request from peripheral ignored

AdrianEggenberger, 2019-11-08:
**Describe the bug**
I'm working with the hci_uart sample with an nRF52 PCA10040 board. I've written a small application that controls the board by exchanging HCI commands over UART.
By sennding the corresponding commands I'm connecting to BLE peripehral device. I'm able to connect and get the expected "LE Enhanced Connection Complete Event" event back.
Also there is a "LE Channel Selection Algorithm Event" received that indicates the used channel selection algorithm.

What is not working is that a connection parameter update request that the peripheral sends (I've checked that with a BLE analyzer) is not indicated by a corresponding HCI event. The connection parameter update request is sillently ignored by the uart_hci device.

**To Reproduce**
Steps to reproduce the behavior:
1. build zephyr sample hci_uart with default settings for the board "nrf52_pca10040"
2. flash firmware to a board
3. send HCI commands to connect to a BLE peripheral
4. force the connected BLE peripheral to send a connection parameter update request
5. -> no HCI event "LE Remote Connection Parameter Request Event" is triggered on the HCI-UART interface

**Expected behavior**
A received connection parameter update request from a connected BLE peripheral should raise a "LE Remote Connection Parameter Request Event" on the HCI interface.

**Impact**
showstopper

**Screenshots or console output**
This is the console output from my PC application that shows the sent and received HCI packages. I've added some lines with "!!!" at the beginning with adescription what happenes in the next section.

!!! The general and LE event masks are configured to receive all LE events
08:28:10,170 DEBUG Executing command line: configure
08:28:10,193  INFO TX ([Command | OGF:0x03/OCF:0x001 | Data:0xFFFFFFFFFF1F0020])
08:28:10,202  INFO RX ([Event | CommandCompleted | Data:0x01010C00])
08:28:10,205  INFO TX ([Command | OGF:0x08/OCF:0x001 | Data:0xFFFF0F0000000000])
08:28:10,210  INFO RX ([Event | CommandCompleted | Data:0x01012000])

!!! Send a connect command to a dedicated device
08:28:12,214 DEBUG Executing command line: connect
08:28:12,225  INFO TX ([Command | OGF:0x08/OCF:0x00D | Data:0xA000400000014806ADA90CF4000C00320000004B000000FFFF])
08:28:12,226  INFO RX ([Event | CommandStatus | Data:0x00010D20])

!!! "LE Enhanced Connection Complete Event" and "LE Channel Selection Algorithm Event"
08:28:12,733  INFO RX ([Event | LeMetaEvent | Data:0x0A00000000014806ADA90CF4000000000000000000000000320000004B0007])
08:28:12,747  INFO RX ([Event | LeMetaEvent | Data:0x14000001])

!!! At this moment the connected peripheral sends a Connection Paramter Update Request, but there is not Event seen on the HCI-UART.

!!! The connection remains active for ~1 minute and is disconnected now.
08:29:10,583 DEBUG Executing command line: disconnect
08:29:10,590  INFO TX ([Command | OGF:0x01/OCF:0x006 | Data:0x000016])
08:29:10,602  INFO RX ([Event | CommandStatus | Data:0x00010604])
08:29:10,612  INFO RX ([Event | DisconnectionComplete | Data:0x00000016])

**Environment (please complete the following information):**
 - Windows 10
 - Segger Embedded Studio 4.20
 - Tag: v2.0.99-ncs1-rc2

jhedberg, 2019-11-08:
Since this is a custom host app, have you made sure that you enable the connection parameter request event using the HCI_LE_Set_Event_Mask command?

jhedberg, 2019-11-08:
> Since this is a custom host app, have you made sure that you enable the connection parameter request event using the HCI_LE_Set_Event_Mask command?

Nevermind - I see "The general and LE event masks are configured to receive all LE events" in your logs.

jhedberg, 2019-11-08:
> 08:28:10,205 INFO TX ([Command | OGF:0x08/OCF:0x001 | Data:0xFFFF0F0000000000])

I'm not sure if that's being decoded correctly or not, but normally what immediately follows 0x.. is the most significant byte, meaning the entire least significant half of the mask is zeroes, which would indicate that you're note enabling the right events. However, considering that you are getting the LE Enhanced Conn Complete I'd conclude that the decoding in your logs is just very confusing.

AdrianEggenberger, 2019-11-08:
At the beginning the invalid event masks caused some issues and I got no event s back at all. I did some try and error tests to find out about the correct bit/byte order, but I think they are correct by now.

AdrianEggenberger, 2019-11-08:
I also did some attempts to debug the issue, but I wasn't sure where the event should come in the system. If I would know where the event should get received I can try to debug it on my side.

carlescufi, 2019-11-08:
Since you are expecting this on the Central side, can you make sure you have `CONFIG_BT_CTLR_CONN_PARAM_REQ=y` in your `.config`?

AdrianEggenberger, 2019-11-09:
To be sure it is enabled I've added "CONFIG_BT_CTLR_CONN_PARAM_REQ=y" to the nrf5.conf file. The behavior is the same after a rebuild. I assume that this feature is enabled by default?

carlescufi, 2019-11-12:
@AdrianEggenberger can you try with the latest master upstream? It looks like you are using Nordic's fork, but this is Zephyr's GitHub bug tracker.

AdrianEggenberger, 2019-11-12:
Yes you are right. My tests were based on the Nordic nRFConnectSDK.

I repeated the tests with the current commit in the official zephyr repository from the 12.11. on 20:37 (c724033accdd6df6579652891bb83ea6a9160a19).

The results are the same.

cvinayak, 2019-11-13:
> !!! At this moment the connected peripheral sends a Connection Paramter Update Request, but there is not Event seen on the HCI-UART.

>!!! The connection remains active for ~1 minute and is disconnected now.
08:29:10,583 DEBUG Executing command line: disconnect
08:29:10,590 INFO TX ([Command | OGF:0x01/OCF:0x006 | Data:0x000016])
08:29:10,602 INFO RX ([Event | CommandStatus | Data:0x00010604])
08:29:10,612 INFO RX ([Event | DisconnectionComplete | Data:0x00000016])

Is the BLE peripheral too a Zephyr sample? Which one?
from your explanation above, the peripheral does not timeout but is been gracefully disconnected after 1 minute by the local HCI-UART based device. This means, the peripheral device did not send any request to HCI_UART device that was not replied to? Do you have logs in the peripheral device?

If you have a spare nRF5 series DK, could you please use the nRF sniffer to capture the Bluetooth trafffic? https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE

The sniffer logs will help quickly identify issues in controller implementation.

AdrianEggenberger, 2019-11-13:
The BLE peripheral is not one of the Zephyr samples. I tried it with one of our own Nordic SDK based peripherals. There I know that they send a connection parameter request ~5s after a connect. If the Zephyr examples have this mechanism as well they could be used too.

The disconnect part on my example can be ignored (in fact I should have removed them from the log) because the disconnect is initiated normally by the HCI command OGF:0x01/OCF:0x006 and handled correctly.

I've no access to our sniffer hardware today, but I'll add a trace tomorrow.
What happens is that the peripheral sends an LL_CONNECTION_PARAM_REQ to the central. As defined in the standard there should be a LL_UNKNOWN_RSP or a LL_CONNECTION_PARAM_REQ as a consequence. But none of them can be seen in the air trace.

On the host interface I would expect that the received LL_CONNECTION_PARAM_REQ is indicated over  the LE Meta Event "LE Remote Connection Parameter Request".

AdrianEggenberger, 2019-11-13:
I was able to get a bit deeper in the issue by a new debugging session.

I found out, that there a an indication on the UART interface when the requests is received, but its not the expected type. 

I expected an LE Meta Event "LE Remote Connection Parameter Request"
(Core Spec 5, Vol 2, Part E, 7.7.65.6
,but what is received is a
"Low Energy L2CAP Signaling" "CONNECTION PARAMETER UPDATE REQUEST"
(Core Spec 5, Vol 3, Part A, 4.20)

What event is the expected one? Or should both be received? The received one somehow makes totally sense as it's obviously received from the remote device over the L2CAP channel 5 ("Low Energy L2CAP Signaling" for LE-U links), but what are the events in the LE  Meta Event section for?

carlescufi, 2019-11-13:
> I expected an LE Meta Event "LE Remote Connection Parameter Request"
> (Core Spec 5, Vol 2, Part E, 7.7.65.6
> ,but what is received is a
> "Low Energy L2CAP Signaling" "CONNECTION PARAMETER UPDATE REQUEST"
> (Core Spec 5, Vol 3, Part A, 4.20)

Both Central and Peripheral must support the Connection Parameters Request Link Layer Control Procedure for the Peripheral to request a conn param update at the Link Layer level. If either of them don't support it the Peripheral will send the L2CAP packet you see.
Most likely your peripheral device does not support the Connection Parameters Request Link Layer Control Procedure and that's why you get the L2CAP packet instead. I understand it's  your Central that you run with Zephyr.

AdrianEggenberger, 2019-11-14:
I was able to record the procedure with the Nordic Sniffer two times (attached)

[ConnParamUpdateReq.zip](https://github.com/zephyrproject-rtos/zephyr/files/3845535/ConnParamUpdateReq.zip)

1) Using Zephyr firmware on the central side and my application controlling it.
2) Using a Silicon Labs BLED Dongle as central controlled by a foreign software.

In both cases the connection parameter request triggered on the peripheral side cases the "Connection Parameter Update Request" in the L2CAP signaling channel. With the silicon labs dongle the request triggers a "Connection Parameter Update Response" on the L2CAP signaling channel followed by a LL_CONNECTION_UPDATE_REQ on the LL.

My assumtion is now, that my software needs to do react on reception of the request on the L2CAP signaling channel. Probably it's now an up to the host controller to decide if the request is accepted or denied an send the corresponding response on the L2CAP signaling channel. So it's probably not a fault in the HCI UART firmware as the L2CAP content is handled transparent on the HCI UART firmware. Right?

carlescufi, 2019-11-14:
> My assumtion is now, that my software needs to do react on reception of the request on the L2CAP signaling channel. Probably it's now an up to the host controller to decide if the request is [...]

That's correct. The handling of the L2CAP packet is up to the Host, which will then need to respond to it with another L2CAP packet (the response) and then initiate an LL Connection Update procedure using the HCI_LE_Connection_Update command.

Closing this as this is not a bug but a misunderstanding of the spec. Feel free to reopen if you disagree.

closed 2019-11-14 by carlescufi

AdrianEggenberger, 2019-11-14:
Thanks for the lesson in the BLE link layer understanding... :-)
I thought that I'm quite familiar with the core spec, but obviously there are still a lot of things to learn.