zephyrproject-rtos/zephyr #20484: Tests/kernel/gen_isr_table failing when enabling WDT driver

albertofloyd, 2019-11-08:
**Describe the bug**
Unable to build MEC15 boards when enabling CONFIG_WATCHDOG=y
There is no other driver that uses same interrupt so suspecting something in the test case.

**To Reproduce**
Steps to reproduce the behavior:
1. Add CONFIG_WATCHDOG=y to  zephyr\boards\arm\mec15xxevb_assy6853\mec15xxevb_assy6853_defconfig
2. cd zephyr/tests/kernel/gen_isr_table/
3. mkdir build; cd build
4. See error

**Expected behavior**
No build error

**Impact**

**Screenshots or console output**
 97%] Generating isr_tables.c
gen_isr_tables.py: multiple registrations at table_index 171 for irq 171 (0xab)
Traceback (most recent call last):
  File "/home/jamezaar/ecwork/zephyr/arch/common/gen_isr_tables.py", line 314, in <module>
    main()
  File "/home/jamezaar/ecwork/zephyr/arch/common/gen_isr_tables.py", line 306, in main
    error("multiple registrations at table_index %d for irq %d (0x%x)" % (table_index, irq, irq))
  File "/home/jamezaar/ecwork/zephyr/arch/common/gen_isr_tables.py", line 36, in error
    raise Exception()

andrewboie, 2019-11-08:
Those hex values for parameter look familiar....the test case is trying to install some dummy IRQs, which are colliding with the real IRQ like for the WDT:

```
FAILED: zephyr/isr_tables.c
cd /home/anashif/zephyrproject/zephyr/build/zephyr && /opt/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --input-target=elf32-littlearm --output-target=binary --only-section=.intList /home/anashif/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.elf isrList.bin && /usr/bin/python3 /home/anashif/zephyrproject/zephyr/arch/common/gen_isr_tables.py --debug --output-source isr_tables.c --kernel /home/anashif/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.elf --intlist isrList.bin   --sw-isr-table --vector-table
gen_isr_tables.py: multiple registrations at table_index 171 for irq 171 (0xab)
gen_isr_tables.py: (174, 0)
gen_isr_tables.py: Configured interrupt routing
gen_isr_tables.py: handler    irq flags param
gen_isr_tables.py: --------------------------
gen_isr_tables.py: 0xe0605    170 0     0xca55e77e
gen_isr_tables.py: 0xe05e1    171 0     0xb01dface
gen_isr_tables.py: 0xe0699    172 1     0x0
gen_isr_tables.py: 0xe0671    173 1     0x0
gen_isr_tables.py: 0xe0cb5    171 0     0x119568
gen_isr_tables.py: 0xe0da1    111 0     0x0
gen_isr_tables.py: 0xe1e5d    17  0     0x1195a4
gen_isr_tables.py: 0xe1e5d    4   0     0x1195b0
gen_isr_tables.py: 0xe1e5d    0   0     0x1195bc
gen_isr_tables.py: 0xe1e5d    1   0     0x1195c8
gen_isr_tables.py: 0xe1e5d    2   0     0x1195d4
gen_isr_tables.py: 0xe1e5d    3   0     0x1195e0
gen_isr_tables.py: 0xe2555    7   0     0x119598
gen_isr_tables.py: 0xe2515    15  0     0x119598
gen_isr_tables.py: 0xe24d5    11  0     0x119598
gen_isr_tables.py: 0xe2b05    101 0     0x1195ec
gen_isr_tables.py: 0xe2b05    100 0     0x1195f8
gen_isr_tables.py: 0xe2c51    135 0     0x0
```

Look here:

```
tests/kernel/gen_isr_table/src/main.c
38-#endif
39-
40:#define ISR3_ARG    0xb01dface
41-#define ISR4_ARG    0xca55e77e
42-#define ISR5_ARG    0xf0ccac1a
```

pabigot, 2019-11-11:
Looks like an instance of #20571.

andrewboie, 2019-11-12:
> Looks like an instance of #20571.

@pabigot it isn't -- in this case, the test case is using an IRQ line that it believes is unused to install a test ISR, but that IRQ line is actually in use.

nashif, 2019-11-12:
does not qualify as medium, enabling hw on a test.

tbursztyka, 2019-12-03:
It's a test logic vs hw irq specs issue: 

On hw config side:
CONFIG_NUM_IRQS 174
And:
DT_INST_0_MICROCHIP_XEC_WATCHDOG_IRQ_0=171

But the test code tries this (line 258):

IRQ_CONNECT(IRQ_LINE(ISR3_OFFSET), 1, isr3, ISR3_ARG, 0);

IRQ_LINE(ISR3_OFFSET) giving also 171

I don't think it's relevant to enabled watchdog for this test (unless you want to detect such collision at built time, which is perfectly done then).

I don't see any bug here.

Shall we close ?

albertofloyd, 2019-12-05:
When I enable the watchdog driver in MEC boards, shippable builds fail, so it can be merged.
Are you indicating this test build conflict can be ignored?

nashif, 2019-12-05:
Maybe just add the board to `platform_exclude` like a few other boards being added there or disable WDT by default on this board and enable it in the application rather in the default config of the baord.

albertofloyd, 2019-12-06:
Will do that

closed 2019-12-06 by albertofloyd