zephyrproject-rtos/zephyr #20489: [Coverity CID :205645]Integer handling issues in /drivers/counter/counter_mchp_xec.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/counter/counter_mchp_xec.c
 Category: Integer handling issues
 Function: counter_xec_set_top_value
 Component: Drivers
 CID: 205645
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
41c93a589c1
```
178static int counter_xec_set_top_value(struct device *dev,
179                                     const struct counter_top_cfg *cfg)
180{
181        BTMR_Type *counter = COUNTER_XEC_REG_BASE(dev);
182        const struct counter_xec_config *counter_cfg = COUNTER_XEC_CONFIG(dev);
183        struct counter_xec_data *data = COUNTER_XEC_DATA(dev);
184        int ret = 0;
185        bool restart;
186
187        if (data->alarm_cb) {
188                return -EBUSY;
189        }
190
191        if (cfg->ticks > counter_cfg->info.max_top_value) {
192                return -EINVAL;
193        }
194
   	
CID 205645 (#1 of 1): Logical vs. bitwise operator (CONSTANT_EXPRESSION_RESULT)
logical_vs_bitwise: The expression counter->CTRL && 32UL is suspicious because it performs a Boolean operation on a constant other than 0 or 1.
   	Did you intend to use & rather than &&?
195        restart = (counter->CTRL && MCHP_BTMR_CTRL_START);
196
197        counter->CTRL &= ~MCHP_BTMR_CTRL_START;
198
199        if (cfg->flags & COUNTER_TOP_CFG_DONT_RESET) {
200                if (counter->CNT > cfg->ticks) {
201                        ret = -ETIME;
202
203                        if (cfg->flags & COUNTER_TOP_CFG_RESET_WHEN_LATE) {
204                                counter->CNT = cfg->ticks;
205                        }
206                }
207        } else {
208                counter->CNT = cfg->ticks;
209        }
```

nashif, 2019-11-27:
@dcpleung  can you please take a look at this one?

PR #21047 (2019-11-28): counter: mchp_xec: fix logical/bit-wise AND
Coverity discovered that a logical AND was used in place of
a bit-wise AND. So fix it.

Fixes #20489
commit 12948fdf (2019-11-28): counter: mchp_xec: fix logical/bit-wise AND

Coverity discovered that a logical AND was used in place of
a bit-wise AND. So fix it.

Fixes #20489
closed 2019-11-28 by nashif