zephyrproject-rtos/zephyr #20490: [Coverity CID :205651]Uninitialized variables in /drivers/dma/dma_stm32.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/dma/dma_stm32.c
 Category: Uninitialized variables
 Function: dma_stm32_configure
 Component: Drivers
 CID: 205651
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
749d2d21bf8

cybertale, 2019-11-26:
@aescolar Is this issue points to `DMA_InitStruct` in `dma_stm32_configure()`? I see this structure do need to be initialized before used. Sorry but I don't see problem description in your comment.

aescolar, 2019-11-26:
@cybertale (I just helped triage it, by trying to pinpointing the origin and author)
This is the whole report from Coverity:
```
231static int dma_stm32_configure(struct device *dev, u32_t id,
232                               struct dma_config *config)
233{
234        struct dma_stm32_data *data = dev->driver_data;
235        struct dma_stm32_stream *stream = &data->streams[id];
236        const struct dma_stm32_config *dev_config =
237                                        dev->config->config_info;
238        DMA_TypeDef *dma = (DMA_TypeDef *)dev_config->base;
   	1. var_decl: Declaring variable DMA_InitStruct without initializer.
239        LL_DMA_InitTypeDef DMA_InitStruct;
240        u32_t msize;
241        int ret;
242
   	2. Condition id >= data->max_streams, taking false branch.
243        if (id >= data->max_streams) {
244                return -EINVAL;
245        }
246
   	3. Condition stream->busy, taking false branch.
247        if (stream->busy) {
248                return -EBUSY;
249        }
250
251        stm32_dma_disable_stream(dma, id);
252        dma_stm32_clear_stream_irq(dev, id);
253
   	4. Condition config->head_block->block_size > 65535, taking false branch.
254        if (config->head_block->block_size > DMA_STM32_MAX_DATA_ITEMS) {
255                LOG_ERR("Data size too big: %d\n",
256                       config->head_block->block_size);
257                return -EINVAL;
258        }
259
   	5. Condition stream->direction == MEMORY_TO_MEMORY, taking true branch.
   	6. Condition !dev_config->support_m2m, taking false branch.
260        if ((stream->direction == MEMORY_TO_MEMORY) &&
261                (!dev_config->support_m2m)) {
262                LOG_ERR("Memcopy not supported for device %s",
263                        dev->config->name);
264                return -ENOTSUP;
265        }
266
   	7. Condition config->source_data_size != 4U, taking true branch.
   	8. Condition config->source_data_size != 2U, taking true branch.
   	9. Condition config->source_data_size != 1U, taking false branch.
267        if (config->source_data_size != 4U &&
268            config->source_data_size != 2U &&
269            config->source_data_size != 1U) {
270                LOG_ERR("Source unit size error, %d",
271                        config->source_data_size);
272                return -EINVAL;
273        }
274
   	10. Condition config->dest_data_size != 4U, taking true branch.
   	11. Condition config->dest_data_size != 2U, taking true branch.
   	12. Condition config->dest_data_size != 1U, taking false branch.
275        if (config->dest_data_size != 4U &&
276            config->dest_data_size != 2U &&
277            config->dest_data_size != 1U) {
278                LOG_ERR("Dest unit size error, %d",
279                        config->dest_data_size);
280                return -EINVAL;
281        }
282
283        /*
284         * STM32's circular mode will auto reset both source address
285         * counter and destination address counter.
[... 123 lines cut ...]
389                DMA_InitStruct.FIFOMode = LL_DMA_FIFOMODE_DISABLE;
390        }
391#endif
392        if (stream->source_periph) {
393                DMA_InitStruct.NbData = config->head_block->block_size /
394                                        config->source_data_size;
395        } else {
396                DMA_InitStruct.NbData = config->head_block->block_size /
397                                        config->dest_data_size;
398        }
399
400        LL_DMA_Init(dma, table_ll_stream[id], &DMA_InitStruct);
401
402        LL_DMA_EnableIT_TC(dma, table_ll_stream[id]);
403
404#if defined(CONFIG_DMA_STM32_V1)
405        if (DMA_InitStruct.FIFOMode == LL_DMA_FIFOMODE_ENABLE) {
406                LL_DMA_EnableIT_FE(dma, table_ll_stream[id]);
407        } else {
408                LL_DMA_DisableIT_FE(dma, table_ll_stream[id]);
409        }
410#endif
411        return ret;
412}
```

dleach02, 2019-12-02:
The specific problem is that the function stm32_dma_check_fifo_mburst() is referencing the FIFOThreshold field of DMA_InitStruct which has never been initialized. I don't know the hardware but I would figure out what FIFOThreshold should be initialized to before calling the check_fifo_mburst() function.

galak, 2019-12-03:
> The specific problem is that the function stm32_dma_check_fifo_mburst() is referencing the FIFOThreshold field of DMA_InitStruct which has never been initialized. I don't know the hardware but I [...]

@erwango any thoughts/comments?

erwango, 2019-12-03:
Shouldn't we do something like :
```
	DMA_InitStruct.FIFOThreshold = stm32_dma_get_fifo_threshold(config->head_block->fifo_mode_control);
```

Seems function `stm32_dma_get_fifo_threshold` is meant to provide threshold value, but return value is not used.

cybertale, 2019-12-04:
> Shouldn't we do something like :
> 
> ```
> DMA_InitStruct.FIFOThreshold = stm32_dma_get_fifo_threshold(config->head_block->fifo_mode_control);
> ```
> 
> Seems function `stm32_dma_get_fifo_threshold` is meant to provide threshold value, but return value is not used.
> 
> Btw, some clean up might be needed as `stm32_dma_enable_fifo` is not used ..
> 
> @cybertale , can you check these points ?

Yes, this is a potential bug I missed, I'll start fixing this now.

PR #21180 (2019-12-09): drivers: dma: stm32: add missing assignment for fifo threshold
Fixes #20490.
commit 65c1d764 (2019-12-09): drivers: dma: stm32: add missing assignment for fifo threshold

Fixes #20490.
closed 2019-12-09 by galak