zephyrproject-rtos/zephyr #20492: [Coverity CID :205653]Control flow issues in /drivers/dma/dma_stm32_v1.c

aasthagr, 2019-11-08:
Static code scan issues seen in File: /drivers/dma/dma_stm32_v1.c
 Category: Control flow issues
 Function: stm32_dma_check_fifo_mburst
 Component: Drivers
 CID: 205653
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
749d2d21bf8
```
287bool stm32_dma_check_fifo_mburst(LL_DMA_InitTypeDef *DMAx)
288{
289        u32_t msize = DMAx->MemoryOrM2MDstDataSize;
290        u32_t fifo_level = DMAx->FIFOThreshold;
291        u32_t mburst = DMAx->MemBurst;
292
293        switch (msize) {
   	
CID 205653 (#1 of 1): Missing break in switch (MISSING_BREAK)
unterminated_case: The case for value 0U is not terminated by a 'break' statement.
294        case LL_DMA_MDATAALIGN_BYTE:
295                switch (mburst) {
296                case LL_DMA_MBURST_INC4:
297                        return true;
298                case LL_DMA_MBURST_INC8:
299                        if (fifo_level == LL_DMA_FIFOTHRESHOLD_1_2 ||
300                            fifo_level == LL_DMA_FIFOTHRESHOLD_FULL) {
301                                return true;
302                        } else {
303                                return false;
304                        }
305                case LL_DMA_MBURST_INC16:
306                        if (fifo_level == LL_DMA_FIFOTHRESHOLD_FULL) {
307                                return true;
308                        } else {
309                                return false;
310                        }
311                }
   	CID 205644: Missing break in switch (MISSING_BREAK) [select issue]
   	fallthrough: The above case falls through to this one.
312        case LL_DMA_MDATAALIGN_HALFWORD:
313                switch (mburst) {
314                case LL_DMA_MBURST_INC4:
315                        if (fifo_level == LL_DMA_FIFOTHRESHOLD_1_2 ||
316                            fifo_level == LL_DMA_FIFOTHRESHOLD_FULL) {
317                                return true;
```

cybertale, 2019-11-26:
This is explained in #20491 .

dleach02, 2019-11-26:
per @cybertale description, this is a false positive and is by design.

nashif, 2020-03-11:
marked as fp, clsoing

closed 2020-03-11 by nashif